tudelft-iv/view-of-delft-dataset #2: k3d unavailable issue

ZixianReid, 2022-05-27:
The 3D figure is not shown. 

run command 

 jupyter nbextension enable  --py --sys-prefix k3d

after 

jupyter nbextension install --py --sys-prefix k3d

BalazsSzekeres, 2022-05-27:
Hi Zixian,

Thank you for raising an issue. Do you mean after applying the two commands the plots are still not displayed?

Could you additionally run `jupyter labextension install k3d`?

ZixianReid, 2022-05-31:
Hi BalazsSzekeres,
     Thanks for the respone. I have additionally run `jupyter labextension install k3d`. And the plots are not available utile I run `jupyter nbextension enable --py --sys-prefix k3d`.

BalazsSzekeres, 2022-05-31:
I understand, I have updated the documentation based on your comment! Thank you for your input!

closed 2022-05-31 by BalazsSzekeres