tudelft-iv/view-of-delft-dataset #3: Data annotation doesn't include object id

LJacksonPan, 2022-05-27:
Hi there,

Firstly great thanks for your dataset which has very good quality radar and LiDAR point clouds. However, we found that your data annotation doesn't include object ids, which are mainly for object tracking tasks. KITTI's tracking label format includes two extra parameters at the beginning for sequence number and tracking id, here's an example:

`0 -1 DontCare -1 -1 -10.000000 219.310000 188.490000 245.500000 218.560000 -1000.000000 -1000.000000 -1000.000000 -10.000000 -1.000000 -1.000000 -1.000000
0 -1 DontCare -1 -1 -10.000000 47.560000 195.280000 115.480000 221.480000 -1000.000000 -1000.000000 -1000.000000 -10.000000 -1.000000 -1.000000 -1.000000
0 0 Van 0 0 -1.793451 296.744956 161.752147 455.226042 292.372804 2.000000 1.823255 4.433886 -4.552284 1.858523 13.410495 -2.115488
0 1 Cyclist 0 0 -1.936993 737.619499 161.531951 931.112229 374.000000 1.739063 0.824591 1.785241 1.640400 1.675660 5.776261 -1.675458
0 2 Pedestrian 0 0 -2.523309 1106.137292 166.576807 1204.470628 323.876144 1.714062 0.767881 0.972283 6.301919 1.652419 8.455685 -1.900245
1 -1 DontCare -1 -1 -10.000000 228.120000 183.030000 258.830000 217.340000 -1000.000000 -1000.000000 -1000.000000 -10.000000 -1.000000 -1.000000 -1.000000
1 -1 DontCare -1 -1 -10.000000 59.210000 191.300000 137.370000 227.430000 -1000.000000 -1000.000000 -1000.000000 -10.000000 -1.000000 -1.000000 -1.000000
1 0 Van 0 0 -1.796862 294.898777 156.024256 452.199718 284.621269 2.000000 1.823255 4.433886 -4.650955 1.766774 13.581085 -2.121565
1 1 Cyclist 0 0 -1.935205 745.017137 156.393157 938.839722 374.000000 1.739063 0.824591 1.785241 1.700640 1.640419 5.778596 -1.664456
1 2 Pedestrian 0 0 -2.530402 1138.342096 160.872449 1223.338201 324.146788 1.714062 0.767881 0.972283 6.352093 1.593046 8.156156 -1.886840
2 -1 DontCare -1 -1 -10.000000 236.270000 175.500000 267.210000 211.030000 -1000.000000 -1000.000000 -1000.000000 -10.000000 -1.000000 -1.000000 -1.000000
2 -1 DontCare -1 -1 -10.000000 68.906000 183.810000 145.870000 224.020000 -1000.000000 -1000.000000 -1000.000000 -10.000000 -1.000000 -1.000000 -1.000000
2 0 Van 0 0 -1.800343 293.093560 150.470149 449.259225 277.104290 2.000000 1.823255 4.433886 -4.749625 1.675025 13.751675 -2.127642
2 1 Cyclist 0 0 -1.933364 752.406083 151.248515 946.562490 374.000000 1.739063 0.824591 1.785241 1.760880 1.605178 5.780931 -1.653453
2 2 Pedestrian 0 0 -2.538744 1151.358043 154.633575 1223.691377 324.375836 1.714062 0.767881 0.972283 6.409693 1.533561 7.859055 -1.873436
3 -1 DontCare -1 -1 -10.000000 243.860000 168.790000 275.270000 204.270000 -1000.000000 -1000.000000 -1000.000000 -10.000000 -1.000000 -1.000000 -1.000000
3 -1 DontCare -1 -1 -10.000000 62.325000 165.380000 166.910000 201.500000 -1000.000000 -1000.000000 -1000.000000 -10.000000 -1.000000 -1.000000 -1.000000
3 0 Van 0 0 -1.803889 291.328036 145.082168 446.400830 269.811515 2.000000 1.823255 4.433886 -4.848295 1.583277 13.922264 -2.133719
3 1 Cyclist 0 0 -1.931469 759.786603 146.098339 954.280160 374.000000 1.739063 0.824591 1.785241 1.821119 1.569936 5.783265 -1.642450
3 2 Pedestrian 0 0 -2.547728 1154.836779 148.360923 1241.000000 321.627088 1.714062 0.767881 0.972283 6.463579 1.474131 7.560739 -1.860031
4 -1 DontCare -1 -1 -10.000000 252.530000 168.660000 284.460000 202.850000 -1000.000000 -1000.000000 -1000.000000 -10.000000 -1.000000 -1.000000 -1.000000
4 -1 DontCare -1 -1 -10.000000 59.320000 170.670000 174.830000 207.910000 -1000.000000 -1000.000000 -1000.000000 -10.000000 -1.000000 -1.000000 -1.000000
4 0 Van 0 0 -1.808333 290.287584 146.641981 444.387179 269.473545 2.000000 1.823255 4.433886 -4.934786 1.601945 14.098646 -2.139796
4 1 Cyclist 0 0 -1.929519 767.158958 140.942948 961.992360 374.000000 1.739063 0.824591 1.785241 1.881359 1.534695 5.785600 -1.631447
4 2 Pedestrian 1 0 -2.557045 1180.675035 151.025283 1241.000000 325.015204 1.714062 0.767881 0.972283 6.516488 1.497786 7.267796 -1.846627`

The KITTI tracking annotation format is specified as following:

[image]

However, I couldn't find any object id annotations. Will the object id annotation be released later? Many thanks in advance!

BalazsSzekeres, 2022-05-28:
Hi!

Thank you for the question! Indeed, it is not included in the current release but is planned to be released later. I will keep this issue open, and close it when it is added, so you will get notified as well.

Hope this answers your question!

andraspalffy, 2022-06-24:
Hi @LJacksonPan, can you contact me privately please? I do have a first version of the trackid-s and I would like to discuss how you would use it!

Thanks, Andras

andraspalffy, 2022-11-30:
Track id-s now are available.

closed 2022-11-30 by andraspalffy

304886938, 2023-02-21:
Hello, thank you very much for providing the dataset containing 4D Radar. I'm trying to do 3D object tracking based on the dataset. I would like to ask if you can provide a tracking evaluation script. I'm looking forward to your reply!

andraspalffy, 2023-02-22:
Hi, we do not have a tracking evaluation script, sorry.
We encourage you to make your own. You can also contact our colleague [Hidde Boekema](https://www.tudelft.nl/staff/h.j.boekema/?cHash=425cdaa1459bd5ce5fa0987ae83d8a1d) as he is working on prediction and probably can direct you a bit better.