tudelft-iv/view-of-delft-dataset #11: Question about BBox annotation.

OuyangJunyuan, 2022-06-30:
[image]
It seems to provide nothing about pitch and roll annotation, right?

andraspalffy, 2022-06-30:
Hello, 

you are right, these angles are not there. This is because the official KITTI annotation format does not support pitch and roll angles, see  [here](https://github.com/bostondiditeam/kitti/blob/master/resources/devkit_object/readme.txt). In other words, we do not have a field to insert them. However, we do have them and plan to share them later.

closed 2022-06-30 by andraspalffy

OuyangJunyuan, 2022-06-30:
> Hello,
>
> you are right, these angles are not there. This is because the official KITTI annotation format does not support pitch and roll angles, see [...]

Awesome! looking forward to the release of pitch and roll.