tudelft-iv/view-of-delft-dataset #16: 1_frame_information.ipynb      helpers.py

hjfenghj, 2022-07-20:
when I try to run the file named "1_frame_information.ipynb";when I use the function named ”vis_3d.draw_plot“ in the part named "LiDAR";I find the problem

TypeError                                 Traceback (most recent call last)
~\AppData\Local\Temp/ipykernel_33332/1146023220.py in <module>
      6 
      7 
----> 8 vis_3d.draw_plot(lidar_origin_plot=True, lidar_points_plot=True)

G:\view-of-delft-dataset-main\vod\visualization\vis_3d.py in draw_plot(self, radar_origin_plot, lidar_origin_plot, camera_origin_plot, lidar_points_plot, radar_points_plot, radar_velocity_plot, annotations_plot, write_to_html, html_name, grid_visible, auto_frame)
     73 
     74         if lidar_origin_plot:
---> 75             self.plot_lidar_origin()
     76 
     77         if camera_origin_plot:

G:\view-of-delft-dataset-main\vod\visualization\vis_3d.py in plot_lidar_origin(self, label, color, axis_length, label_size)
    121     def plot_lidar_origin(self,label = True,color = lidar_plot_color_3d,axis_length = axis_length_3d,label_size = axis_label_size):
    122 
--> 123         self.plot += k3d_get_axes(self.transform_matrices['lidar'], axis_length)
    124         if label:
    125             self.plot += k3d.text("lidar",

G:\view-of-delft-dataset-main\vod\visualization\helpers.py in k3d_get_axes(hom_transform_matrix, axis_length)
    183                         dtype=float)
    184 
--> 185     origin = [hom_transform_matrix[:3, 3]] * 3
    186     axis = hom_transform_matrix.dot(hom_vector.T).T[:, :3]
    187 

TypeError: 'method' object is not subscriptable

andraspalffy, 2022-07-20:
Hello, can you please copy the exact codeblock you used (above and beyond the questioned line), using the code block formatting here so it is easy to read? 

And please do the same for the error message, it is really hard to read it like this.
Alternatively, you could also add screenshots!

thanks

hjfenghj, 2022-07-20:
I am so sorry，I have find  this problem is caused by my adding comments；So，I have solve it；I have closed the issue now;
I'm sorry for your precious time

closed 2022-07-20 by hjfenghj

andraspalffy, 2022-07-20:
No worries, glad to hear that it works for you! :)

hjfenghj, 2022-10-11:
I am so sorry，I have find  this problem is caused by my adding comments；So，I have solve it；I have closed the issue now;
I'm sorry for your precious time

 
| |
冯浩俊
|
|
***@***.***
|