tudelft-iv/view-of-delft-dataset #17: pp-radar performance on validation set

gc625, 2022-07-20:
Hi, I was wondering what were the performance results for pp-radar on the validation set of the data? When and how will we be able to test our own custom models on the test set? Thank you!

gc625, 2022-07-20:
I see, thank you. Regarding pointpillar performance on the validation set, would it be possible to release your results?

andraspalffy, 2022-08-01:
Sorry, we do not have that easily available for the final splits. Maybe later we can release that too.

TarunKumar1995-glitch, 2022-08-03:
The model config shared in the pp-radar [guidelines](https://github.com/tudelft-iv/view-of-delft-dataset/blob/main/PP-Radar.md) was an outdated version. It is now updated and should provide comparable results.

closed 2022-11-01 by andraspalffy