tudelft-iv/view-of-delft-dataset #18: PP-Radar class car perform abnormal

Chen-hai-feng, 2022-07-26:
Hi,when I run pp-radar copy from your pp-radar guidelines[https://github.com/tudelft-iv/view-of-delft-dataset/blob/main/PP-Radar.md](url).
The output is as follows:
`INFO  Car AP@0.70, 0.70, 0.70:
**bbox AP:22.6309, 28.9965, 23.7317
bev  AP:31.4834, 38.0852, 31.8317
3d   AP:5.7727, 11.2009, 10.5092
aos  AP:21.41, 27.94, 23.06
Car AP_R40@0.70, 0.70, 0.70:
bbox AP:18.3606, 25.2401, 21.1341
bev  AP:28.1924, 34.1668, 29.1861
3d   AP:4.2473, 8.4565, 7.0809
aos  AP:17.11, 24.20, 20.25
Car AP@0.70, 0.50, 0.50:
bbox AP:22.6309, 28.9965, 23.7317
bev  AP:42.1046, 48.2684, 40.7786
3d   AP:28.5084, 32.6852, 30.5675
aos  AP:21.41, 27.94, 23.06
Car AP_R40@0.70, 0.50, 0.50:
bbox AP:18.3606, 25.2401, 21.1341
bev  AP:40.2857, 44.8250, 37.9114
3d   AP:24.4887, 31.3992, 26.7819
aos  AP:17.11, 24.20, 20.25**
Pedestrian AP@0.50, 0.50, 0.50:
bbox AP:27.3352, 25.2310, 23.8031
bev  AP:21.9817, 19.3415, 18.4007
3d   AP:12.0408, 11.6275, 11.0378
aos  AP:26.68, 24.45, 23.02
Pedestrian AP_R40@0.50, 0.50, 0.50:
bbox AP:24.2204, 21.3113, 19.0969
bev  AP:17.3610, 15.1017, 13.3590
3d   AP:5.4015, 4.7779, 4.1870
aos  AP:23.44, 20.46, 18.31
Pedestrian AP@0.50, 0.25, 0.25:
bbox AP:27.3352, 25.2310, 23.8031
bev  AP:43.4798, 38.3300, 36.0452
3d   AP:35.8189, 30.6979, 28.5039
aos  AP:26.68, 24.45, 23.02
Pedestrian AP_R40@0.50, 0.25, 0.25:
bbox AP:24.2204, 21.3113, 19.0969
bev  AP:41.6760, 36.1811, 32.1889
3d   AP:33.1692, 27.9692, 24.4298
aos  AP:23.44, 20.46, 18.31
Cyclist AP@0.50, 0.50, 0.50:
bbox AP:54.0451, 49.8941, 43.8147
bev  AP:56.1162, 50.8601, 45.0722
3d   AP:24.1656, 22.1448, 19.4740
aos  AP:53.70, 49.62, 43.58
Cyclist AP_R40@0.50, 0.50, 0.50:
bbox AP:53.7608, 48.1834, 42.7750
bev  AP:54.6610, 48.8692, 44.0406
3d   AP:22.8138, 20.1624, 17.6555
aos  AP:53.42, 47.87, 42.49
Cyclist AP@0.50, 0.25, 0.25:
bbox AP:54.0451, 49.8941, 43.8147
bev  AP:76.6063, 70.5171, 63.4638
3d   AP:68.7943, 62.0618, 55.2496
aos  AP:53.70, 49.62, 43.58
Cyclist AP_R40@0.50, 0.25, 0.25:
bbox AP:53.7608, 48.1834, 42.7750
bev  AP:77.1777, 70.6806, 63.6326
3d   AP:68.7145, 61.9188, 55.3979
aos  AP:53.42, 47.87, 42.49
`

you can see that，class Car **easy perform well than moderate**，i do not know how to deal with this problem.
Could you please tell me why it happens...  or can you give some guideline

andraspalffy, 2022-07-26:
Sorry, I am not sure if I got your problem. However, please use our eval.
https://github.com/tudelft-iv/view-of-delft-dataset/blob/main/4_evaluation.ipynb

Chen-hai-feng, 2022-07-26:
> Sorry, I am not sure if I got your problem. However, please use our eval. https://github.com/tudelft-iv/view-of-delft-dataset/blob/main/4_evaluation.ipynb

Thank you.I also have one question as follows:

This config file is copy from your PP-Radar.md  named Pointpillars Config .
```
CLASS_NAMES: ['Car', 'Pedestrian', 'Cyclist']

DATA_CONFIG:
    _BASE_CONFIG_: cfgs/dataset_configs/radar_5frames_as_kitti_dataset.yaml
    POINT_CLOUD_RANGE: [0, -25.6, -3, 51.2, 25.6, 2]
    DATA_PROCESSOR:
        - NAME: mask_points_and_boxes_outside_range
          REMOVE_OUTSIDE_BOXES: True

        - NAME: shuffle_points
          SHUFFLE_ENABLED: {
            'train': True,
            'test': False
          }

        - NAME: transform_points_to_voxels
          VOXEL_SIZE: [0.16, 0.16, 5]
          MAX_POINTS_PER_VOXEL: 10
          MAX_NUMBER_OF_VOXELS: {
            'train': 16000,
            'test': 40000
          }
    DATA_AUGMENTOR:
        DISABLE_AUG_LIST: ['random_world_rotation', 'gt_sampling']
        AUG_CONFIG_LIST:

            - NAME: random_world_flip
              ALONG_AXIS_LIST: ['x']

            - NAME: random_world_scaling
              WORLD_SCALE_RANGE: [0.95, 1.05]

MODEL:
    NAME: PointPillar

    VFE:
        NAME: RadarPillarVFE
        WITH_DISTANCE: False
        USE_ABSLOTE_XYZ: True
        USE_RCS: True
        USE_ELEVATION: True
        USE_VR: True
        USE_VXYZ: False
        USE_TIME: True
        USE_NORM: True
        NUM_FILTERS: [64]

    MAP_TO_BEV:
        NAME: PointPillarScatter
        NUM_BEV_FEATURES: 64

    BACKBONE_2D:
        NAME: BaseBEVBackbone
        LAYER_NUMS: [3, 5, 5]
        LAYER_STRIDES: [2, 2, 2]
        NUM_FILTERS: [64, 128, 256]
        UPSAMPLE_STRIDES: [1, 2, 4]
        NUM_UPSAMPLE_FILTERS: [128, 128, 128]

    DENSE_HEAD:
        NAME: AnchorHeadSingle
        CLASS_AGNOSTIC: False

        USE_DIRECTION_CLASSIFIER: True
        DIR_OFFSET: 0.78539
        DIR_LIMIT_OFFSET: 0.0
        NUM_DIR_BINS: 2

        ANCHOR_GENERATOR_CONFIG: [
            {
                'class_name': 'Car',
                'anchor_sizes': [[3.9, 1.6, 1.56]],
                'anchor_rotations': [0, 1.57],
                'anchor_bottom_heights': [-1.78],
                'align_center': False,
                'feature_map_stride': 2,
                'matched_threshold': 0.6,
                'unmatched_threshold': 0.45
            },
            {
                'class_name': 'Pedestrian',
                'anchor_sizes': [[0.8, 0.6, 1.73]],
                'anchor_rotations': [0, 1.57],
                'anchor_bottom_heights': [-0.6],
                'align_center': False,
                'feature_map_stride': 2,
                'matched_threshold': 0.5,
                'unmatched_threshold': 0.35
            },
            {
                'class_name': 'Cyclist',
                'anchor_sizes': [[1.76, 0.6, 1.73]],
                'anchor_rotations': [0, 1.57],
                'anchor_bottom_heights': [-0.6],
                'align_center': False,
                'feature_map_stride': 2,
                'matched_threshold': 0.5,
                'unmatched_threshold': 0.35
            }
        ]

        TARGET_ASSIGNER_CONFIG:
            NAME: AxisAlignedTargetAssigner
            POS_FRACTION: -1.0
            SAMPLE_SIZE: 512
            NORM_BY_NUM_EXAMPLES: False
            MATCH_HEIGHT: False
            BOX_CODER: ResidualCoder

        LOSS_CONFIG:
            LOSS_WEIGHTS: {
                'cls_weight': 1.0,
                'loc_weight': 2.0,
                'dir_weight': 0.2,
                'code_weights': [1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0]
            }

    POST_PROCESSING:
        RECALL_THRESH_LIST: [0.3, 0.5, 0.7]
        SCORE_THRESH: 0.1
        OUTPUT_RAW_SCORE: False

        EVAL_METRIC: kitti

        NMS_CONFIG:
            MULTI_CLASSES_NMS: False
            NMS_TYPE: nms_gpu
            NMS_THRESH: 0.01
            NMS_PRE_MAXSIZE: 4096
            NMS_POST_MAXSIZE: 500

OPTIMIZATION:
    BATCH_SIZE_PER_GPU: 16
    NUM_EPOCHS: 80

    OPTIMIZER: adam_onecycle
    LR: 0.003
    WEIGHT_DECAY: 0.01
    MOMENTUM: 0.9

    MOMS: [0.95, 0.85]
    PCT_START: 0.4
    DIV_FACTOR: 10
    DECAY_STEP_LIST: [35, 45]
    LR_DECAY: 0.1
    LR_CLIP: 0.0000001

    LR_WARMUP: False
    WARMUP_EPOCH: 1

    GRAD_NORM_CLIP: 10

```
The   `EVAL_METRIC: kitti` means it uses kitti eval. Openpcdet offers the kitti eval code but not perform well on class Car.
So should I change the official Openpcdet kitti eval code to your [eval](https://github.com/tudelft-iv/view-of-delft-dataset/blob/main/4_evaluation.ipynb)?

andraspalffy, 2022-07-26:
No, the eval code, as the example files show, take a list of txt files, each of which is a set of detections/groundtruth labels.

To use it, you should:

1. train your model as you wish
2. eval it on your eval set
3. fetch the results and turn them to txt-s, see for example here: [](https://github.com/open-mmlab/OpenPCDet/issues/1031)
4. run our eval.

This is btw how you have to submit results to KITTI benchmark as well.

Chen-hai-feng, 2022-07-31:
> No, the eval code, as the example files show, take a list of txt files, each of which is a set of detections/groundtruth labels.
> 
> To use it, you should:
> 
> 1. train your model as you wish
> 2. eval it on your eval set
> 3. fetch the results and turn them to txt-s, see for example here: [How to make the test result of KITTI test dataset into txt open-mmlab/OpenPCDet#1031](https://github.com/open-mmlab/OpenPCDet/issues/1031)
> 4. run our eval.
> 
> This is btw how you have to submit results to KITTI benchmark as well.

Hi,really thanks.
I do some evaluation according to your instruction above.The result of `pp-radar(x,y,z,RCS,vr)` :
```
Evaluating kitti by default
mAP Image BBox finished
mAP bev BBox finished
mAP 3D BBox finished
Evaluating kitti by ROI
mAP Image BBox finished
mAP bev BBox finished
mAP 3D BBox finished
Results: 
Entire annotated area: 
Car: 30.19642913369689 
Pedestrian: 13.974993487887472 
Cyclist: 37.506525529095015 
mAP: 27.225982716893128 
Driving corridor area: 
Car: 68.96345935323946 
Pedestrian: 20.268735890890298 
Cyclist: 72.04015535536024 
mAP: 53.75745019983 

```
You can find the `Pedestrian`'s score is too low.in your paper 20-30 nearlly,but it is only 14.
This config file is copy from your PP-Radar.md ,and dataset config also.
Could you please tell me how to reproduce your performance on your paper？

andraspalffy, 2022-08-01:
Did you follow the exact steps, including all modifications from the manual? E.g. VFE, kitti config, kitti_dataset.py, etc?

Also, note that the results in the paper are on the test set, training on train+val. 
Further, as said many times before, this is a dataset repository, not a novel code line being shared to be reproduced.

Chen-hai-feng, 2022-08-01:
> Did you follow the exact steps, including all modifications from the manual? E.g. VFE, kitti config, kitti_dataset.py, etc?
>
> Also, note that the results in the paper are on the test set, training on [...]

Thank you for sparing  time to answering my question.

Yes.I have modified all related files as VFE，kitti_dataset,and kitti config according to `PP-Radar.md` in this repository.The reason why i want to reproduce this results is that we mark your results as baseline and find some ways to promote performance.

I find `PP-radar.md`  use val test to evaluate as
```
DATA_SPLIT: {
    'train': train,
    'test': val
}

INFO_PATH: {
    'train': [kitti_infos_train.pkl],
    'test': [kitti_infos_val.pkl],
}
```
Also, I train the model base on the train set and test on the val set, because the test set don‘t have groundtruth` label_2`.I am curious about how to get testing set label or can I divide train&val set to train&val&test？
And do you test pp-radar on val test, I really confused about why my result is too low and pedestrian also.

andraspalffy, 2022-08-01:
Hello!

We will not release the test labels, just like KITTI or Nuscenes do not.
Instead, you will be able to upload your detections of the test set (with a detector trained on train+val), and we will run it on the test set automatically. This is coming in the next 3 weeks.

 @TarunKumar1995-glitch will have a look at the performance on val set, but it is not a priority for now.

andraspalffy, 2022-08-01:
Please continue in #17

closed 2022-08-01 by andraspalffy

min-zhang, 2023-07-05:
> 3\. 如何将KITTI测试数据集的测试结果转为txt

Hello, how to convert result to txt result and how to use https://github.com/tudelft-iv/view-of-delft-dataset/blob/main/4_evaluation.ipynb
2023-07-05 14:16:18,853   INFO  Car AP@0.70, 0.70, 0.70:
bbox AP:25.2931, 35.8688, 30.2304
bev  AP:34.3534, 41.2079, 37.0070
3d   AP:13.5977, 17.5404, 14.6623
aos  AP:22.00, 34.01, 28.79
Car AP_R40@0.70, 0.70, 0.70:
bbox AP:21.6496, 32.8158, 26.9882
bev  AP:31.7833, 40.4042, 33.8964
3d   AP:7.0489, 12.1813, 9.6848
aos  AP:17.99, 30.88, 25.39
Car AP@0.70, 0.50, 0.50:
bbox AP:25.2931, 35.8688, 30.2304
bev  AP:53.9248, 58.3119, 50.3598
3d   AP:34.7278, 41.2159, 36.9743
aos  AP:22.00, 34.01, 28.79
Car AP_R40@0.70, 0.50, 0.50:
bbox AP:21.6496, 32.8158, 26.9882
bev  AP:52.7421, 57.3406, 49.3869
3d   AP:31.6778, 40.5127, 33.8923
aos  AP:17.99, 30.88, 25.39
Pedestrian AP@0.50, 0.50, 0.50:
bbox AP:28.7664, 26.7176, 22.2800
bev  AP:26.4072, 21.3220, 20.4711
3d   AP:16.2190, 13.5128, 13.1590
aos  AP:28.29, 26.09, 21.88
Pedestrian AP_R40@0.50, 0.50, 0.50:
bbox AP:26.2926, 21.6179, 19.5190
bev  AP:22.2260, 17.9081, 16.0608
3d   AP:9.9817, 8.1055, 6.9494
aos  AP:25.61, 20.92, 18.87
Pedestrian AP@0.50, 0.25, 0.25:
bbox AP:28.7664, 26.7176, 22.2800
bev  AP:48.2595, 40.4788, 33.9854
3d   AP:38.4618, 31.6497, 30.0935
aos  AP:28.29, 26.09, 21.88
Pedestrian AP_R40@0.50, 0.25, 0.25:
bbox AP:26.2926, 21.6179, 19.5190
bev  AP:46.2196, 37.6173, 33.2550
3d   AP:36.3881, 30.1446, 26.2724
aos  AP:25.61, 20.92, 18.87
Cyclist AP@0.50, 0.50, 0.50:
bbox AP:60.9057, 58.7827, 52.4480
bev  AP:70.8855, 63.8257, 56.6782
3d   AP:36.7310, 31.9714, 31.0531
aos  AP:60.82, 58.68, 52.36
Cyclist AP_R40@0.50, 0.50, 0.50:
bbox AP:62.0980, 56.1747, 50.1514
bev  AP:71.2272, 63.1126, 56.4553
3d   AP:33.3966, 29.5327, 26.2988
aos  AP:62.00, 56.07, 50.06
Cyclist AP@0.50, 0.25, 0.25:
bbox AP:60.9057, 58.7827, 52.4480
bev  AP:84.8932, 76.9164, 68.6398
3d   AP:81.7922, 73.7306, 66.0486
aos  AP:60.82, 58.68, 52.36
Cyclist AP_R40@0.50, 0.25, 0.25:
bbox AP:62.0980, 56.1747, 50.1514
bev  AP:86.0499, 78.9152, 71.5783
3d   AP:81.5084, 72.7171, 65.6249
aos  AP:62.00, 56.07, 50.06