tudelft-iv/view-of-delft-dataset #25: Questions about the labels.

ChenJcumtb, 2022-08-28:
Hi Mr. Palffy, I was wondering if you could share with me your opinion on some details regarding the specification of the labeling? When I checked the labels of the dataset, I found that Rider and Cyclist are present at the same time, in addition to the presence of the Bicycle label. This was rather surprising to me. I am looking forward to your response, which will help me a lot. Best wishes! ChenJ.

andraspalffy, 2022-08-28:
Hello, thanks for the question, and I can see why is it not trivial. Please refer to:

[https://tudelft-iv.github.io/view-of-delft-dataset/docs/ANNOTATION.html#annotated-classes](https://tudelft-iv.github.io/view-of-delft-dataset/docs/ANNOTATION.html#annotated-classes)

To conclude:

- A "Cyclist" is a bicycle being ridden, with its rider. Its bounding box includes both the ride and the rider.
- A "Rider" is a human riding a bicycle, motor, wheelchair, etc. vehicle. The bounding box only includes the human. This annotation is useful for pose estimation tasks for example. You can ignore it if you are doing cyclist detection.
- A "Bicycle" annotation is an unused bicycle, of which we have several here in the Netherlands. You can also ignore these if you are doing cyclist detection, but it is useful for researchers interested in the diverse appearance of the bicycle class itself.

closed 2022-08-28 by andraspalffy

ChenJcumtb, 2022-08-29:
> Hello, thanks for the question, and I can see why is it not trivial. Please refer to:
>
> https://tudelft-iv.github.io/view-of-delft-dataset/docs/ANNOTATION.html#annotated-classes
>
> To conclude:
>
> * A [...]

Thank you for your reply. Yes, I agree with you that the Bicycle category is very supportive of the study. I will make better use of this dataset work in my subsequent research, thank you for your patience. ChenJ.

andraspalffy, 2022-08-29:
Glad to hear! Please let me know if you have further questions.

Andras

ChenJcumtb, 2022-08-30:
> Glad to hear! Please let me know if you have further questions.
> 
> Andras

I have no further questions at the moment. Your answer has dissipated a dark cloud in my head while analyzing the data. Thank you very much for your patient reply. Our work is ongoing. I look forward to still having such pleasant discussions as we are having now in the latter part of our work. ChenJ.