tudelft-iv/view-of-delft-dataset #27: mmdetection3d Adaptation check

code88881234, 2022-11-01:
Dear author, thank you for your meaningful work！
We adapt vod to mmdet3d framework. In addition, eval was modified according to official instructions to evaluate the Entire annotated area. We want to check whether this result is similar to the val set result of 1296 frames in your framework, because we want to see the baseline result of the val set. Thank you and look forward to your reply.
[image]

closed 2022-11-01 by andraspalffy

vehicle-jyw, 2022-12-24:
@code88881234 hello，i am also doing some work by mmdet3d, would you share your  contact information, i want ask a few questions, thank you!

code88881234, 2022-12-24:
> @code88881234 hello，i am also doing some work by mmdet3d, would you share your contact information, i want ask a few questions, thank you!

OK，give me your wechat

vehicle-jyw, 2022-12-24:
@code88881234 17768727368

JZhong-Li, 2023-03-02:
@code88881234 hello, Thanks for your shared results. I also modified the eval.py file in mmdet3d and trained some basic model utilizing pp-radar data, but the evaluation results were abnormal. Thus, I would like to ask what kind of pp-radar you used, 1-, 3-, or 5-scan. In addition, I would appreciate it greatly if you could share your modified evaluation files with me. Thanks.

code88881234, 2023-03-13:
> @code88881234 hello, Thanks for your shared results. I also modified the eval.py file in mmdet3d and trained some basic model utilizing pp-radar data, but the evaluation results were abnormal. Thus, [...]
You should use the official script to evaluate the results and rewrite the official evaluation script into eval in mmdet3d. The evaluation still needs to be based on the results of each frame detection according to the official script, and we have adjusted the results so far.

JZhong-Li, 2023-03-13:
> 

Thanks for your reply! I have modified the eval.py according to the official script, and applied the pointpillar to the data of pp-radar (single scan). However, the results were still terrible, especially for pedestrian and car. Are there any other files that need to be modified? I noticed that the classes' order in mmdet3d is ['Pedestrian', 'Cyclist', 'Car'], while ['Car', 'Cyclist', 'Pedestrian'] in openPCDet. Whether this change will make a terrible result? From this viewpoint, I modified the order of these classes in both files of data creation and config. But, the results for Car and Pedestrian becomes zero. Could you please give me any suggestion? Thanks a lot.

JZhong-Li, 2023-03-13:
@code88881234 Thanks for your reply! I have modified the eval.py according to the official script, and applied the pointpillar to the data of pp-radar (single scan). However, the results were still terrible, especially for pedestrian and car. Are there any other files that need to be modified? I noticed that the classes' order in mmdet3d is ['Pedestrian', 'Cyclist', 'Car'], while ['Car', 'Cyclist', 'Pedestrian'] in openPCDet. Whether this change will make a terrible result? From this viewpoint, I modified the order of these classes in both files of data creation and config. But, the results for Car and Pedestrian becomes zero. Could you please give me any suggestion? Thanks a lot.

Lime2333, 2023-11-17:
> > @code88881234 hello，i am also doing some work by mmdet3d, would you share your contact information, i want ask a few questions, thank you!
> 
> OK，give me your wechat

@code88881234 Hello, when I use the dataset to mmdet3d, I encountered a few problems. So may I ask which code file you modified when using it? And is it convenient for you to share your config files with me? Best wishes！
My email：1043275367@qq.com