tudelft-iv/view-of-delft-dataset #28: PP radar issue while training with openpcdet

GaneshAdam, 2022-11-28:
I have followed steps mentioned in "https://github.com/tudelft-iv/view-of-delft-dataset/blob/main/PP-Radar.md".
However I face below error when running train.py

022-11-28 15:19:03,527   INFO  cfg.OPTIMIZATION.LR_CLIP: 1e-07
2022-11-28 15:19:03,527   INFO  cfg.OPTIMIZATION.LR_WARMUP: False
2022-11-28 15:19:03,527   INFO  cfg.OPTIMIZATION.WARMUP_EPOCH: 1
2022-11-28 15:19:03,527   INFO  cfg.OPTIMIZATION.GRAD_NORM_CLIP: 10
2022-11-28 15:19:03,527   INFO  cfg.TAG: pointpillar_radar
2022-11-28 15:19:03,527   INFO  cfg.EXP_GROUP_PATH: kitti_models
2022-11-28 15:19:03,547   INFO  Loading KITTI dataset
2022-11-28 15:19:03,718   INFO  Total samples for KITTI dataset: 5139
[320 320   1]
Traceback (most recent call last):
  File "train.py", line 223, in <module>
    main()
  File "train.py", line 125, in main
    model = build_network(model_cfg=cfg.MODEL, num_class=len(cfg.CLASS_NAMES), dataset=train_set)
  File "../pcdet/models/__init__.py", line 18, in build_network
    model_cfg=model_cfg, num_class=num_class, dataset=dataset
  File "../pcdet/models/detectors/__init__.py", line 37, in build_detector
    model_cfg=model_cfg, num_class=num_class, dataset=dataset
  File "../pcdet/models/detectors/pointpillar.py", line 7, in __init__
    self.module_list = self.build_networks()
  File "../pcdet/models/detectors/detector3d_template.py", line 47, in build_networks
    model_info_dict=model_info_dict
  File "../pcdet/models/detectors/detector3d_template.py", line 81, in build_vfe
    depth_downsample_factor=model_info_dict['depth_downsample_factor']
TypeError: __init__() got an unexpected keyword argument 'grid_size'

Did anyone faced similar issue and have solution?

andraspalffy, 2022-11-28:
Sorry, I cannot help from the top of my head. 

Can you maybe format this as a code sample and/or past screenshots of the relevant parts?
This downsample_factor seems like a new param we didn't have in our implementation, maybe we forked sooner.
@TarunKumar1995-glitch  thoughts?

GaneshAdam, 2022-11-29:
Hi andraspalffy,

Thank you for your quick response. The issue is resolved!
I was trying to run PP-radar with latest master of Openpcdet.
I could resolve the issue by comparing the version of Openpcdet you used for PP-radar(https://github.com/open-mmlab/OpenPCDet/tree/0642cf06d0fd84f50cc4c6c01ea28edbc72ea810) with master and did changes in my master version.

closed 2022-11-30 by andraspalffy

mendoza-G, 2023-02-09:
> Hi andraspalffy,
>
> Thank you for your quick response. The issue is resolved! I was trying to run PP-radar with latest master of Openpcdet. I could resolve the issue by comparing the version of [...]

I' m facing the same problem, could you please share the specific solution worked, thanks! the OpenPCDet 0.3(recommended), whiched I 've tried, facing the issue of  spconv library, it seems spconv 1.x is needed, any suggestions? Looking forward to your reply~

mendoza-G, 2023-02-11:
> I' m facing the same problem, could you please share the specific solution worked, thanks! the OpenPCDet 0.3(recommended), whiched I 've tried, facing the issue of spconv library, it seems spconv [...]

Thanks a lot, the problem has been solved.

taowucheng1026, 2023-12-21:
> Hi andraspalffy,
>
> Thank you for your quick response. The issue is resolved! I was trying to run PP-radar with latest master of Openpcdet. I could resolve the issue by comparing the version of [...]

I' m facing the same problem, could you please share the specific solution worked, thank you very much!