tudelft-iv/view-of-delft-dataset #29: problems in mmdet3d training.

vehicle-jyw, 2022-12-17:
Thank you all for your execllent dataset!
I've seen previous issues about low results, and i know you used openpcd, I modified the config files of mmdet3d according to the previous issues(including point range, voxel sizes and so on). however, **I still got very low val accuracy even ‘0’!!!**(i already comented the eval line).
I really don't know how to solve this problem, and I know you don't provide tools to answer questions, so I can only venture to ask you for help, thank you all!!!
(this picture is fisrt epoch)
[image]

andraspalffy, 2022-12-22:
As discussed before and in with emails with you, we do not provide support for your implementations for object detection.

Some clear issues that I can see here though, and can help others:

- voxel is taller than the point cloud range height you use
- max number of points per voxel is set to 10, too low for LiDAR
- you use the built in eval tool, not ours.

Hope these helps, also have a look please at previous issues.

Good luck!

closed 2022-12-22 by andraspalffy

JZhong-Li, 2023-03-13:
@vehicle-jyw hello, I also applied this dataset to mmdetection3d. Have you solved this issue?

Lime2333, 2023-11-21:
@LiuLiu-55 @JZhong-Li hello,when I use the dataset to mmdet3d, I encountered a few problems. Have you solved this issue?

LiuLiu-55, 2023-11-23:
> @LiuLiu-55 @JZhong-Li hello,when I use the dataset to mmdet3d, I encountered a few problems. Have you solved this issue?
I use the eval tool they provide not the OpenPCDet, and the performance gets better.

Lime2333, 2023-11-27:
> > @LiuLiu-55 @JZhong-Li hello,when I use the dataset to mmdet3d, I encountered a few problems. Have you solved this issue?
> > I use the eval tool they provide not the OpenPCDet, and the performance gets better.

@LiuLiu-55 Then do you finally reappear at the results of the Pointpillars reappeared at MMDET3D.