tudelft-iv/view-of-delft-dataset #30: Errors at 2D annotations

santimontiel, 2022-12-22:
Good morning everyone.

First of all, thank you for your incredible dataset. Deep Radar Perception can continue its development thanks to dataset like this one.

I was exploring the dataset and training a YOLO v5 model to take advantage of the camera images. I have observed that there is an error that occurs systematically and makes the detector's performance decrease. There are images in which a road user is annotated within the full image (tending to be 100% width and 100% height).

Here you can have an example at 01064.txt.

`Car 908 2 -2.6248431176513627 **0.0 0.0 1935.0 1215.0** 1.9223383609753752 2.0535622747106395 4.999146108042289 2.9836005613724135 1.6872684689437945 2.159001243918928 -1.6804516475522036 1
`
Personally, I am working with classes Pedestrian, Cyclist and Car and I have found 183 of these issues in the dataset. If you want more information, I am open to collaborate through this issue.

Best regards.

andraspalffy, 2022-12-22:
Dear @santimontiel , thanks, nice spot!

We calculate these 2D boxes by projecting the 3D bounding boxes to the camera image. Probably in these cases something went off, and default values were used. @srimannarayanabaratam can you have a look?

santimontiel, 2023-01-11:
Hello everyone!

I would like to know if there is any advance on this topic.

Is there any reference on performance of 2D detectors on this dataset?

Best regards.

srimannarayanabaratam, 2023-01-11:
Hi @santimontiel! Unfortunately, we have not had time to look into this issue yet. We hope to fix this issue within this month.

srimannarayanabaratam, 2023-03-18:
@santimontiel The error seems to have been rooted in a PCDet utility library used to project 3D bounding boxes to 2D images (`box_utils.boxes3d_kitti_camera_to_imageboxes`).

Such cases occur in rare boundary conditions where part of the 3D bounding box falls behind the camera centre. We would recommend removing/ignoring such boxes while training/validating your 2D image model.

closed 2023-03-18 by srimannarayanabaratam