tudelft-iv/view-of-delft-dataset #33: The model of pp-radar (5 scans) is not evaluated well

JZhong-Li, 2023-02-16:
I have trained the radar_5frames_pointpillars.yaml based on the , but the evaluated results are not good. The 'Truncation' in eval.py has also been commented out. Here is my results and config files.

[image]

[image]

[image]

[image]

JZhong-Li, 2023-02-16:
Sorry, something is missing. The backbone is RadarPillarVFE.

andraspalffy, 2023-02-16:
Hello. Some things to notice here:
1, you use the OpenPCDet eval kit, not ours. That is ok if that is what you want, but you have to be very sure which number means what.
2, You use both VR and VR_comp. Why? Just VR is just noise if you have the compensated.
3, Maybe you forgot to normalize your data during loading.

Hope this helps! 
Andras

closed 2023-02-16 by andraspalffy

shuxiusuxiu, 2023-09-09:
> Hello. Some things to notice here: 1, you use the OpenPCDet eval kit, not ours. That is ok if that is what you want, but you have to be very sure which number means what. 2, You use both VR and [...]

Hi! i met the same problem, the result seems not satisfactory, and you suggested to normalize data.
But i have some confusions about normalization.

1) what is the motivation of normalizing point?
2) which attributes should be normalized? (v_comp, RCS?)
3) the statictics(mean, std) used for normalization should be computed in the whole traning set(one for all the frame) or just in current frame

andraspalffy, 2023-09-11:
> 

see #69