tudelft-iv/view-of-delft-dataset #34: sensors

555time, 2023-02-17:
hello,I want to know the height of the lidar, camera, and radar above the ground, and the distance between the lidar and radar in the front and back.

andraspalffy, 2023-02-17:
Hello, thank you for the question!
I calculated transformations for you compared to the center of front left wheel. You can assume a normal wheel size for a Toyota Prius 2013. As you can see, LiDAR is ~1.35 m, Radar is ~0.2 m, camera is ~0.8 m above the center of the wheels. (note, camera frame is upside down). I also added LiDAR to radar for you, although this is available via the toolkit as well (by going from LiDAR to camera, and camera to Radar).

Hope this helps! 
p.s. I don not know what you mean by front and back.

LiDAR to front_left_wheel

```
Translation: [1.967, 0.750, -1.350]
Rotation: in Quaternion [0.000, 0.000, 0.704, 0.710]
in RPY (radian) [0.000, -0.000, 1.562]
in RPY (degree) [0.000, -0.000, 89.501]
```

Radar to front_left_wheel

```
Translation: [-0.547, 0.693, -0.195]
Rotation: in Quaternion [0.003, 0.004, 0.702, 0.712]
in RPY (radian) [0.010, 0.002, 1.556]
in RPY (degree) [0.572, 0.100, 89.152]
```

Camera to front_left_wheel
```
Translation: [-0.636, 1.101, 0.871]
Rotation: in Quaternion [0.006, -0.663, 0.748, -0.005]
in RPY (radian) [-1.451, -0.003, -3.126]
in RPY (degree) [-83.137, -0.154, -179.095]
```

LiDAR to radar:
```
Translation: [2.516, 0.060, -1.151]
Rotation: in Quaternion [0.001, -0.005, 0.003, 1.000]
in RPY (radian) [0.002, -0.010, 0.006]
in RPY (degree) [0.091, -0.573, 0.349]
```

closed 2023-02-17 by andraspalffy

dcasadoherr, 2023-07-06:
Hello!
Would be useful to know the distance to the rear wheel axle as well :)
Thanks in advance!

andraspalffy, 2023-07-12:
@dcasadoherr sorry for delayed answer, we are on holiday.

Camera to rear_left_wheel:
```
- Translation: [-0.664, 0.759, -1.968]
- Rotation: in Quaternion [0.006, -0.663, 0.748, -0.005]
            in RPY (radian) [-1.451, -0.003, -3.126]
            in RPY (degree) [-83.137, -0.154, -179.095]
```