tudelft-iv/view-of-delft-dataset #44: ValueError: need at least one array to stack

Lime2333, 2023-04-23:
Hello, Author,  thank you for your codes.

After I get the training results with OpenPCdet, how can I test them?

it means :
How should I set "test_annotation_file" and "result_path" in the vod.evaluate?
[image]

and the dataset and output are as follow:
[image]
[image]

When I set them as shown above, the following error will appear.
```
Evaluating kitti by default
Traceback (most recent call last):
  File "/public/user_data/openpc/OpenPCDet/VoD-main/vod/evaluation/evaluate.py", line 75, in <module>
    main()
  File "/public/user_data/openpc/OpenPCDet/VoD-main/vod/evaluation/evaluate.py", line 57, in main
    current_class=[0, 1, 2])
  File "/public/user_data/openpc/OpenPCDet/VoD-main/vod/evaluation/evaluate.py", line 45, in evaluate
    evaluation_result.update(get_official_eval_result(gt_annotations, dt_annotations, current_class, custom_method=0))
  File "/public/user_data/openpc/OpenPCDet/VoD-main/vod/evaluation/kitti_official_evaluate.py", line 743, in get_official_eval_result
    custom_method=custom_method)
  File "/public/user_data/openpc/OpenPCDet/VoD-main/vod/evaluation/kitti_official_evaluate.py", line 648, in do_eval
    min_overlaps, compute_aos, custom_method=custom_method)
  File "/public/user_data/openpc/OpenPCDet/VoD-main/vod/evaluation/kitti_official_evaluate.py", line 533, in eval_class
    rets = calculate_iou_partly(dt_annotations, gt_annotations, metric, num_parts)
  File "/public/user_data/openpc/OpenPCDet/VoD-main/vod/evaluation/kitti_official_evaluate.py", line 403, in calculate_iou_partly
    total_dt_num = np.stack([len(a["name"]) for a in dt_annotations], 0)
  File "<__array_function__ internals>", line 6, in stack
  File "/home/ubuntu/anaconda3/envs/lib/python3.7/site-packages/numpy/core/shape_base.py", line 422, in stack
    raise ValueError('need at least one array to stack')
ValueError: need at least one array to stack
```

Lime2333, 2023-04-23:
Ok, I have finished the test by debugging and other means, but the test results are quite different from the papers. I wonder if I entered the wrong file.
[image]

the test results:

```
Results: 
Entire annotated area: 
Car: 32.41045092594593 
Pedestrian: 15.382081422159152 
Cyclist: 27.334938159405674 
mAP: 25.042490169170254 
Driving corridor area: 
Car: 69.07015854027503 
Pedestrian: 20.57044282173243 
Cyclist: 40.52262274307669 
mAP: 43.38774136836138 

```

and the result in the paper should been:
[image]

andraspalffy, 2023-04-25:
Hello, since your original problem has been solved, I  will close this issue.
About your eval results, this could be influenced by many things, for example maybe you trained with only one scan of data, not 5, or made a mistake in the training/testing config.

closed 2023-04-25 by andraspalffy

arkadiy-telegin, 2023-06-26:
I'm having the same issue.

@Lime2333 could you please provide the code you found that works? I put the equivalent paths as in your second post, but still getting the same error. Do I need to copy or move some files?

@andraspalffy would be great if the evaluation code is more documented. So far the interface is very opaque and requires reverse engineering to understand what parameters need to be passed.

andraspalffy, 2023-06-26:
Hey @arkadiy-telegin can you be more specific what part you find "very opaque"? Large chunk of the code is from here:

[https://github.com/traveller59/kitti-object-eval-python](https://github.com/traveller59/kitti-object-eval-python
)

Which is not over documented for sure.

min-zhang, 2023-12-22:
@Lime2333 from vod.evaluation import evaluation_common as kitti
from .kitti_official_evaluate import get_official_eval_result
from vod import get_frame_list_from_folder

class Evaluation:
    """
    Evaluation class for KITTI evaluation.
    """
    def __init__(self, test_annotation_file):
        """
        Initialize the evaluation class for KITTI evaluation.
        :param test_annotation_file: Location of the test annotation files.
        """
        self.test_annotation_file = test_annotation_file

    def evaluate(self,
                 result_path,
                 current_class=None,
                 score_thresh=-1,
                 ):
        """
        Evaluate the results.
        :param result_path: Detection labels path.
        :param current_class: Class to evaluate.
        :param score_thresh: Score threshold to use.
        :return: Results of the evaluation.
        """

        if current_class is None:
            current_class = [0, 1, 2]

        val_image_ids = get_frame_list_from_folder(result_path)

        dt_annotations = kitti.get_label_annotations(result_path, val_image_ids)

        if score_thresh > 0:
            dt_annotations = kitti.filter_annotations_low_score(dt_annotations, score_thresh)

        gt_annotations = kitti.get_label_annotations(self.test_annotation_file, val_image_ids)

        evaluation_result = {}
        evaluation_result.update(get_official_eval_result(gt_annotations, dt_annotations, current_class, custom_method=0))
        evaluation_result.update(get_official_eval_result(gt_annotations, dt_annotations, current_class, custom_method=3))

        return evaluation_result

if __name__ == '__main__':

    path_label = '/media/Data/zm/second1/15xiugai/OpenPCDet/data/radar_5frames/training/label_2'
    path_results = 'edia/Data/zm/second1/15xiugai/OpenPCDet/output/kitti_models/pointpillarsradar/default/eval/epoch_80/val/default/result.pkl'

    evaluation = Evaluation(test_annotation_file=path_label)

    results = evaluation.evaluate(
        result_path=path_results,
        current_class=[0, 1, 2])

    print("Results: \n"
          f"Entire annotated area: \n"
          f"Car: {results['entire_area']['Car_3d_all']} \n"
          f"Pedestrian: {results['entire_area']['Pedestrian_3d_all']} \n"
          f"Cyclist: {results['entire_area']['Cyclist_3d_all']} \n"
          f"mAP: {(results['entire_area']['Car_3d_all'] + results['entire_area']['Pedestrian_3d_all'] + results['entire_area']['Cyclist_3d_all']) / 3} \n"
          f"Driving corridor area: \n"
          f"Car: {results['roi']['Car_3d_all']} \n"
          f"Pedestrian: {results['roi']['Pedestrian_3d_all']} \n"
          f"Cyclist: {results['roi']['Cyclist_3d_all']} \n"
          f"mAP: {(results['roi']['Car_3d_all'] + results['roi']['Pedestrian_3d_all'] + results['roi']['Cyclist_3d_all']) / 3} \n"
          ) 
Here's my code, but every time I get the following error：
Traceback (most recent call last):
  File "/media/Data/zm/second1/15xiugai/OpenPCDet/vod/evaluation/evaluate.py", line 3, in <module>
    from .kitti_official_evaluate import get_official_eval_result
ImportError: attempted relative import with no known parent package.

min-zhang, 2023-12-22:
@andraspalffy from vod.evaluation import evaluation_common as kitti
from .kitti_official_evaluate import get_official_eval_result
from vod import get_frame_list_from_folder

class Evaluation:
"""
Evaluation class for KITTI evaluation.
"""
def init(self, test_annotation_file):
"""
Initialize the evaluation class for KITTI evaluation.
:param test_annotation_file: Location of the test annotation files.
"""
self.test_annotation_file = test_annotation_file

def evaluate(self,
             result_path,
             current_class=None,
             score_thresh=-1,
             ):
    """
    Evaluate the results.
    :param result_path: Detection labels path.
    :param current_class: Class to evaluate.
    :param score_thresh: Score threshold to use.
    :return: Results of the evaluation.
    """

    if current_class is None:
        current_class = [0, 1, 2]

    val_image_ids = get_frame_list_from_folder(result_path)

    dt_annotations = kitti.get_label_annotations(result_path, val_image_ids)

    if score_thresh > 0:
        dt_annotations = kitti.filter_annotations_low_score(dt_annotations, score_thresh)

    gt_annotations = kitti.get_label_annotations(self.test_annotation_file, val_image_ids)

    evaluation_result = {}
    evaluation_result.update(get_official_eval_result(gt_annotations, dt_annotations, current_class, custom_method=0))
    evaluation_result.update(get_official_eval_result(gt_annotations, dt_annotations, current_class, custom_method=3))

    return evaluation_result
if name == 'main':

path_label = '/media/Data/zm/second1/15xiugai/OpenPCDet/data/radar_5frames/training/label_2'
path_results = 'edia/Data/zm/second1/15xiugai/OpenPCDet/output/kitti_models/pointpillarsradar/default/eval/epoch_80/val/default/result.pkl'

evaluation = Evaluation(test_annotation_file=path_label)

results = evaluation.evaluate(
    result_path=path_results,
    current_class=[0, 1, 2])

print("Results: \n"
      f"Entire annotated area: \n"
      f"Car: {results['entire_area']['Car_3d_all']} \n"
      f"Pedestrian: {results['entire_area']['Pedestrian_3d_all']} \n"
      f"Cyclist: {results['entire_area']['Cyclist_3d_all']} \n"
      f"mAP: {(results['entire_area']['Car_3d_all'] + results['entire_area']['Pedestrian_3d_all'] + results['entire_area']['Cyclist_3d_all']) / 3} \n"
      f"Driving corridor area: \n"
      f"Car: {results['roi']['Car_3d_all']} \n"
      f"Pedestrian: {results['roi']['Pedestrian_3d_all']} \n"
      f"Cyclist: {results['roi']['Cyclist_3d_all']} \n"
      f"mAP: {(results['roi']['Car_3d_all'] + results['roi']['Pedestrian_3d_all'] + results['roi']['Cyclist_3d_all']) / 3} \n"
      ) 
Here's my code, but every time I get the following error：
Traceback (most recent call last):
File "/media/Data/zm/second1/15xiugai/OpenPCDet/vod/evaluation/evaluate.py", line 3, in
from .kitti_official_evaluate import get_official_eval_result
ImportError: attempted relative import with no known parent package.

Petros626, 2025-04-15:
> I'm having the same issue.
>
> [@Lime2333](https://github.com/Lime2333) could you please provide the code you found that works? I put the equivalent paths as in your second post, but still getting the [...]

I compared the two images and see only changed paths, nothing changed in code.