tudelft-iv/view-of-delft-dataset #45: Result of mAOS

Lime2333, 2023-05-04:
As you said, the evaluation results of dataset need to use your function [vod.evaluate](https://github.com/tudelft-iv/view-of-delft-dataset/blob/main/vod/evaluation/evaluate.py).

However, the output of this function does not include mAOS. 
[image]

How should I get this result?

SantiMontiel, 2023-05-11:
Hi @Lime2333! I'm trying to obtain results of PP-radar (5 frames) and I'm getting a 0 score on cyclists and pedestrians. Could you please share how did you adapt OpenPCDet to Radar?

Thank you so much in advance!

Xliu1009, 2023-05-12:
> Hi @Lime2333! I'm trying to obtain results of PP-radar (5 frames) and I'm getting a 0 score on cyclists and pedestrians. Could you please share how did you adapt OpenPCDet to Radar?
> 
> Thank you so much in advance!
I encountered the same issue and can confirm that it's not the evaluation function, but rather the trained model producing few predictions with low confidence.

Xliu1009, 2023-05-12:
> Hi @Lime2333! I'm trying to obtain results of PP-radar (5 frames) and I'm getting a 0 score on cyclists and pedestrians. Could you please share how did you adapt OpenPCDet to Radar?
> 
> Thank you so much in advance!

I am facing the same problem and looking forward to a reply

Xliu1009, 2023-05-12:
> Hi @Lime2333! I'm trying to obtain results of PP-radar (5 frames) and I'm getting a 0 score on cyclists and pedestrians. Could you please share how did you adapt OpenPCDet to Radar?
> 
> Thank you so much in advance!

I am facing the same problem and looking forward to a reply.

Lime2333, 2023-05-16:
I adapted it according to this issue.
#8 
Maybe this can help you.
@SantiMontiel

closed 2023-05-16 by Lime2333

Lime2333, 2023-05-16:
Remember to use the evaluation function provided by the author to evaluate at last.

And that result of mAOS can also be output by modifing the evaluation function. For example:
`
    print("Results: \n"
          f"Entire annotated area: \n"
          f"Car: {results['entire_area']['Car_3d_all']} \n"
          f"Car_AOS: {results['entire_area']['Car_aos_all']} \n"
          f"Pedestrian: {results['entire_area']['Pedestrian_3d_all']} \n"
          f"Pedestrian_AOS: {results['entire_area']['Pedestrian_aos_all']} \n"
          f"Cyclist: {results['entire_area']['Cyclist_3d_all']} \n"
          f"Cyclist_AOS: {results['entire_area']['Cyclist_aos_all']} \n"
          f"mAP: {(results['entire_area']['Car_3d_all'] + results['entire_area']['Pedestrian_3d_all'] + results['entire_area']['Cyclist_3d_all']) / 3} \n"
          f"mAOS: {(results['entire_area']['Car_aos_all'] + results['entire_area']['Pedestrian_aos_all'] + results['entire_area']['Cyclist_aos_all']) / 3} \n"
          f"Driving corridor area: \n"
          f"Car: {results['roi']['Car_3d_all']} \n"
          f"Car_AOS: {results['roi']['Car_aos_all']} \n"
          f"Pedestrian: {results['roi']['Pedestrian_3d_all']} \n"
          f"Pedestrian_AOS: {results['roi']['Pedestrian_aos_all']} \n"
          f"Cyclist: {results['roi']['Cyclist_3d_all']} \n"
          f"Cyclist_AOS: {results['roi']['Cyclist_aos_all']} \n"
          f"mAP: {(results['roi']['Car_3d_all'] + results['roi']['Pedestrian_3d_all'] + results['roi']['Cyclist_3d_all']) / 3} \n"
          f"mAOS: {(results['roi']['Car_aos_all'] + results['roi']['Pedestrian_aos_all'] + results['roi']['Cyclist_aos_all']) / 3} \n"
          )

`