tudelft-iv/view-of-delft-dataset #47: Ego vehicle velocity

amvishnu67, 2023-05-29:
Hello!
I was trying to find the ego vehicle speed for each frame, for eg: from the odometry data! I couldn't fnd anything and I would like to know if I should estimate the vehicle speed from the radar data? Also, I need the exact vehilce velocity in each frame, not the radial one (But I can do the  transformation if only radial velocity is available) so that I could comapare it with similar frames on nuscenes! Please let me know how to proceed on this.

andraspalffy, 2023-06-01:
Hello, I suggest to take the odom to camera transformations and calculate the smoothed velocity from those. As frames are by 10 Hz, this will give you a good idea of the movement of the car. I.e. the car will move x, y, z meters during 0.1 seconds.

closed 2023-06-01 by andraspalffy

amvishnu67, 2023-10-10:
Hi,
Thanks for the update. I have done as you suggested and now I was wondering
if I should do the same method for finding the speed of the annotations. I
know the radar data has the absolute speed of the data points, but I
clearly need the speed of pedestrian and bike annotations. Please let me
know if you already have a method to get the speed of the annotations or
else I would have to do the estimation procedure!
Thanks and Regards,
Vishnu Vijayan

andraspalffy, 2023-10-11:
Hi, we dont have a ready to use method for this, suggest to have a look at our other publication "hidden gems" github.