tudelft-iv/view-of-delft-dataset #52: 车的检测精度还可以，但是行人和自行车的检测精度很低

min-zhang, 2023-06-14:


Yinbao-Li, 2023-06-20:
您可以看看历史问题，有人此前遇到了相同的问题，按照已关闭的问题中提供的解决方案进行操作即可，希望帮到您。

andraspalffy, 2023-06-21:
Guys please talk in English, or talk privately.
Thank you

closed 2023-06-21 by andraspalffy

min-zhang, 2023-07-05:
Hello, my test result is as follows, epoch is 80,
INFO  Car AP@0.70, 0.70, 0.70:
bbox AP:36.2793, 39.2191, 32.0413
bev  AP:42.8272, 47.2691, 40.7447
3d   AP:17.6040, 20.9249, 19.2227
aos  AP:34.15, 38.25, 31.37
Car AP_R40@0.70, 0.70, 0.70:
bbox AP:33.1363, 37.1293, 30.5799
bev  AP:40.0895, 45.7915, 37.5233
3d   AP:11.6239, 17.0281, 13.8623
aos  AP:30.95, 36.06, 29.66
Car AP@0.70, 0.50, 0.50:
bbox AP:36.2793, 39.2191, 32.0413
bev  AP:63.7014, 60.0651, 51.6864
3d   AP:43.5480, 46.8279, 40.3347
aos  AP:34.15, 38.25, 31.37
Car AP_R40@0.70, 0.50, 0.50:
bbox AP:33.1363, 37.1293, 30.5799
bev  AP:62.9937, 59.0316, 51.2108
3d   AP:41.3002, 44.1800, 37.1804
aos  AP:30.95, 36.06, 29.66
Pedestrian AP@0.50, 0.50, 0.50:
bbox AP:0.0845, 0.1779, 0.1848
bev  AP:0.0567, 0.0942, 0.0942
3d   AP:0.0301, 0.0501, 0.0501
aos  AP:0.06, 0.09, 0.09
Pedestrian AP_R40@0.50, 0.50, 0.50:
bbox AP:0.0449, 0.0896, 0.0930
bev  AP:0.0156, 0.0259, 0.0259
3d   AP:0.0083, 0.0138, 0.0138
aos  AP:0.03, 0.04, 0.04
Pedestrian AP@0.50, 0.25, 0.25:
bbox AP:0.0845, 0.1779, 0.1848
bev  AP:0.1384, 0.3047, 0.2964
3d   AP:0.1028, 0.2191, 0.2133
aos  AP:0.06, 0.09, 0.09
Pedestrian AP_R40@0.50, 0.25, 0.25:
bbox AP:0.0449, 0.0896, 0.0930
bev  AP:0.0680, 0.1462, 0.1439
3d   AP:0.0540, 0.1018, 0.1002
aos  AP:0.03, 0.04, 0.04
Cyclist AP@0.50, 0.50, 0.50:
bbox AP:0.0403, 0.0403, 0.0403
bev  AP:0.0000, 0.0000, 0.0000
3d   AP:0.0000, 0.0000, 0.0000
aos  AP:0.01, 0.01, 0.01
Cyclist AP_R40@0.50, 0.50, 0.50:
bbox AP:0.0333, 0.0333, 0.0333
bev  AP:0.0000, 0.0000, 0.0000
3d   AP:0.0000, 0.0000, 0.0000
aos  AP:0.01, 0.01, 0.01
Cyclist AP@0.50, 0.25, 0.25:
bbox AP:0.0403, 0.0403, 0.0403
bev  AP:0.1310, 0.1310, 0.1310
3d   AP:0.1198, 0.1198, 0.1198
aos  AP:0.01, 0.01, 0.01
Cyclist AP_R40@0.50, 0.25, 0.25:
bbox AP:0.0333, 0.0333, 0.0333
bev  AP:0.0728, 0.0728, 0.0728
3d   AP:0.0662, 0.0662, 0.0662
aos  AP:0.01, 0.01, 0.01
the detection accuracy of people and bicycles is very low, I modified it by your https://github.com/tudelft-iv/view-of-delft-dataset/blob/main/PP-Radar.md, and the modified file is as follows:
radar_5frames_as_kitti_dataset.yaml
`DATASET: 'KittiDataset'
DATA_PATH: '../data/radar_5frames'

POINT_CLOUD_RANGE: [0, -25.6, -3, 51.2, 25.6, 2]

DATA_SPLIT: {
    'train': train,
[... 479 lines cut ...]
    'MeanVFE': MeanVFE,
    'PillarVFE': PillarVFE,
    'Radar7PillarVFE': Radar7PillarVFE,
    'ImageVFE': ImageVFE,
    'DynMeanVFE': DynamicMeanVFE,
    'DynPillarVFE': DynamicPillarVFE,
    'DynamicPillarVFESimple2D': DynamicPillarVFESimple2D
}
`
pcdet/datasets/kitti/kitti_dataset.py
`def get_lidar(self, idx):
        lidar_file = self.root_split_path / 'velodyne' / ('%s.bin' % idx)
        assert lidar_file.exists()
        number_of_channels = 7  # ['x', 'y', 'z', 'rcs', 'v_r', 'v_r_comp', 'time']
        points = np.fromfile(str(lidar_file), dtype=np.float32).reshape(-1, number_of_channels)
        # replace the list values with statistical values; for x, y, z and time, use 0 and 1 as means and std to avoid normalization
        means = [0, 0, 0, 0, 0, 0, 0]  # 'x', 'y', 'z', 'rcs', 'v_r', 'v_r_comp', 'time'
        stds = [1, 1, 1, 1, 1, 1, 1]  # 'x', 'y', 'z', 'rcs', 'v_r', 'v_r_comp', 'time'

        # we then norm the channels
        points = (points - means) / stds
        return points
        # return np.fromfile(str(lidar_file), dtype=np.float32).reshape(-1, number_of_channels)
        # return np.fromfile(str(lidar_file), dtype=np.float32).reshape(-1, 4)`
Can you help me see what is the reason

min-zhang, 2023-07-05:
> Guys please talk in English, or talk privately. Thank you
Hello, my test result is as follows, epoch is 80,
INFO Car AP@0.70, 0.70, 0.70:
bbox AP:36.2793, 39.2191, 32.0413
bev AP:42.8272, 47.2691, 40.7447
3d AP:17.6040, 20.9249, 19.2227
aos AP:34.15, 38.25, 31.37
Car AP_R40@0.70, 0.70, 0.70:
bbox AP:33.1363, 37.1293, 30.5799
bev AP:40.0895, 45.7915, 37.5233
3d AP:11.6239, 17.0281, 13.8623
aos AP:30.95, 36.06, 29.66
Car AP@0.70, 0.50, 0.50:
bbox AP:36.2793, 39.2191, 32.0413
bev AP:63.7014, 60.0651, 51.6864
3d AP:43.5480, 46.8279, 40.3347
aos AP:34.15, 38.25, 31.37
Car AP_R40@0.70, 0.50, 0.50:
bbox AP:33.1363, 37.1293, 30.5799
bev AP:62.9937, 59.0316, 51.2108
3d AP:41.3002, 44.1800, 37.1804
aos AP:30.95, 36.06, 29.66
Pedestrian AP@0.50, 0.50, 0.50:
bbox AP:0.0845, 0.1779, 0.1848
bev AP:0.0567, 0.0942, 0.0942
3d AP:0.0301, 0.0501, 0.0501
aos AP:0.06, 0.09, 0.09
Pedestrian AP_R40@0.50, 0.50, 0.50:
bbox AP:0.0449, 0.0896, 0.0930
bev AP:0.0156, 0.0259, 0.0259
3d AP:0.0083, 0.0138, 0.0138
aos AP:0.03, 0.04, 0.04
Pedestrian AP@0.50, 0.25, 0.25:
bbox AP:0.0845, 0.1779, 0.1848
bev AP:0.1384, 0.3047, 0.2964
3d AP:0.1028, 0.2191, 0.2133
aos AP:0.06, 0.09, 0.09
Pedestrian AP_R40@0.50, 0.25, 0.25:
bbox AP:0.0449, 0.0896, 0.0930
bev AP:0.0680, 0.1462, 0.1439
3d AP:0.0540, 0.1018, 0.1002
aos AP:0.03, 0.04, 0.04
Cyclist AP@0.50, 0.50, 0.50:
bbox AP:0.0403, 0.0403, 0.0403
bev AP:0.0000, 0.0000, 0.0000
3d AP:0.0000, 0.0000, 0.0000
aos AP:0.01, 0.01, 0.01
Cyclist AP_R40@0.50, 0.50, 0.50:
bbox AP:0.0333, 0.0333, 0.0333
bev AP:0.0000, 0.0000, 0.0000
3d AP:0.0000, 0.0000, 0.0000
aos AP:0.01, 0.01, 0.01
Cyclist AP@0.50, 0.25, 0.25:
bbox AP:0.0403, 0.0403, 0.0403
bev AP:0.1310, 0.1310, 0.1310
3d AP:0.1198, 0.1198, 0.1198
aos AP:0.01, 0.01, 0.01
Cyclist AP_R40@0.50, 0.25, 0.25:
bbox AP:0.0333, 0.0333, 0.0333
bev AP:0.0728, 0.0728, 0.0728
3d AP:0.0662, 0.0662, 0.0662
aos AP:0.01, 0.01, 0.01
the detection accuracy of people and bicycles is very low, I modified it by your https://github.com/tudelft-iv/view-of-delft-dataset/blob/main/PP-Radar.md, and the modified file is as follows:
radar_5frames_as_kitti_dataset.yaml
`DATASET: 'KittiDataset'
DATA_PATH: '../data/radar_5frames'

POINT_CLOUD_RANGE: [0, -25.6, -3, 51.2, 25.6, 2]

DATA_SPLIT: {
[... 473 lines cut ...]
'VFETemplate': VFETemplate,
'MeanVFE': MeanVFE,
'PillarVFE': PillarVFE,
'Radar7PillarVFE': Radar7PillarVFE,
'ImageVFE': ImageVFE,
'DynMeanVFE': DynamicMeanVFE,
'DynPillarVFE': DynamicPillarVFE,
'DynamicPillarVFESimple2D': DynamicPillarVFESimple2D
}
pcdet/datasets/kitti/kitti_dataset.pydef get_lidar(self, idx):
lidar_file = self.root_split_path / 'velodyne' / ('%s.bin' % idx)
assert lidar_file.exists()
number_of_channels = 7 # ['x', 'y', 'z', 'rcs', 'v_r', 'v_r_comp', 'time']
points = np.fromfile(str(lidar_file), dtype=np.float32).reshape(-1, number_of_channels)
# replace the list values with statistical values; for x, y, z and time, use 0 and 1 as means and std to avoid normalization
means = [0, 0, 0, 0, 0, 0, 0] # 'x', 'y', 'z', 'rcs', 'v_r', 'v_r_comp', 'time'
stds = [1, 1, 1, 1, 1, 1, 1] # 'x', 'y', 'z', 'rcs', 'v_r', 'v_r_comp', 'time'

    # we then norm the channels
    points = (points - means) / stds
    return points
    # return np.fromfile(str(lidar_file), dtype=np.float32).reshape(-1, number_of_channels)
    # return np.fromfile(str(lidar_file), dtype=np.float32).reshape(-1, 4)`

Can you help me see what is the reason