tudelft-iv/view-of-delft-dataset #54: Question regarding reproducing paper results

Alexmsit, 2023-06-30:
Hi everyone,

First of all, thank you for publishing this great dataset. Right now i am working on reproducing the paper results.
I followed the instructions within PP-Radar.md and get nearly the same mAP-scores as reported in the paper.
But the classwise AP-scores seem to differ from the paper-results to some extent.

Here is the comparison of the results for training of PP on single-frame radar data with all 7 features used:

[image]

The first line shows the paper results and the second line shows the reproduction results.
Now my question is why the classwise AP-scores differ up to ~10% from the paper results.
Did you use any additional configuration like a weighted loss-function or other things which would explain this difference, or am i missing something here?

Thanks in advance!
Best regards,

Alex

yangyt37, 2023-07-11:
May I ask how you can obtain the results for pedestrians and cyclists? The map I obtained using the official evaluation indicators of openpcdet is less than 1. My result is in the following tabel.

andraspalffy, 2023-07-12:
@Alexmsit thanks for your interest. 10% difference is unfortunately plausible given the dataset size.
Also I guess you are using the validation set, feel free to send it for a test set eval.

@yangyt37 your results suggest that you use a wrong evaluation approach. See this comment which points out where was this answered previously.

https://github.com/tudelft-iv/view-of-delft-dataset/issues/8#issuecomment-1231418778

closed 2023-07-12 by andraspalffy

yangyt37, 2023-07-13:
I use right evaluation approach and I got the result like this:
Car: 32.760911122320536 
Pedestrian: 27.134859105528225 
Cyclist: 59.60893396401781 
mAP: 39.83490139728885 
Driving corridor area: 
Car: 68.9206407782023 
Pedestrian: 36.17506242356443 
Cyclist: 80.8428254073373 
mAP: 61.979509536368006

min-zhang, 2023-08-09:
> I use right evaluation approach and I got the result like this: Car: 32.760911122320536 Pedestrian: 27.134859105528225 Cyclist: 59.60893396401781 mAP: 39.83490139728885 Driving corridor area: Car: 68.9206407782023 Pedestrian: 36.17506242356443 Cyclist: 80.8428254073373 mAP: 61.979509536368006

you can provide me with the dataset, and I can help you with the calculations.Four sets of data in total.
1.
INFO  Car AP@0.70, 0.70, 0.70:
bbox AP:29.2754, 37.7937, 31.9468
bev  AP:38.4106, 46.0381, 39.4115
3d   AP:16.5922, 20.8887, 17.5489
aos  AP:25.90, 36.16, 30.63
Car AP_R40@0.70, 0.70, 0.70:
bbox AP:26.3670, 34.6987, 29.4077
bev  AP:35.5528, 43.5953, 36.7469
3d   AP:11.8746, 16.0044, 12.7312
aos  AP:22.75, 33.00, 27.84
Car AP@0.70, 0.50, 0.50:
bbox AP:29.2754, 37.7937, 31.9468
bev  AP:55.4508, 57.8449, 50.1450
3d   AP:39.2968, 46.3682, 39.5512
aos  AP:25.90, 36.16, 30.63
Car AP_R40@0.70, 0.50, 0.50:
bbox AP:26.3670, 34.6987, 29.4077
bev  AP:54.1486, 57.2201, 48.2378
3d   AP:36.6693, 43.8487, 36.9838
aos  AP:22.75, 33.00, 27.84
Pedestrian AP@0.50, 0.50, 0.50:
bbox AP:28.9486, 27.1331, 26.0896
bev  AP:26.8249, 21.8494, 20.8565
3d   AP:16.8029, 14.0226, 13.2201
aos  AP:28.50, 27.83, 24.77
Pedestrian AP_R40@0.50, 0.50, 0.50:
bbox AP:26.4394, 22.8352, 20.6211
bev  AP:23.0455, 18.6491, 16.5344
3d   AP:11.3486, 9.1428, 7.9505
aos  AP:25.77, 22.05, 19.89
Pedestrian AP@0.50, 0.25, 0.25:
bbox AP:28.6086, 27.1331, 26.0896
bev  AP:48.5453, 41.1766, 39.4566
3d   AP:39.2645, 36.5330, 30.8388
aos  AP:29.50, 26.83, 24.77
Pedestrian AP_R40@0.50, 0.25, 0.25:
bbox AP:26.4394, 22.8352, 20.6211
bev  AP:47.7989, 39.7481, 35.1812
3d   AP:38.2997, 32.1196, 28.0489
aos  AP:25.77, 22.05, 19.89
Cyclist AP@0.50, 0.50, 0.50:
bbox AP:63.7330, 61.2801, 54.3684
bev  AP:71.2002, 63.0600, 57.1477
3d   AP:39.1116, 37.5996, 32.8188
aos  AP:63.59, 61.12, 54.23
Cyclist AP_R40@0.50, 0.50, 0.50:
bbox AP:65.6098, 59.1844, 52.9149
bev  AP:71.2757, 64.1361, 57.6604
3d   AP:36.5646, 33.4715, 29.9994
aos  AP:65.46, 59.03, 52.78
Cyclist AP@0.50, 0.25, 0.25:
bbox AP:63.7330, 61.2801, 54.3684
bev  AP:85.9964, 77.8251, 69.9316
3d   AP:85.4105, 73.1937, 67.3817
aos  AP:63.59, 61.12, 54.23
Cyclist AP_R40@0.50, 0.25, 0.25:
bbox AP:65.6098, 59.1844, 52.9149
bev  AP:86.9980, 79.1704, 72.0699
3d   AP:81.9620, 72.9490, 66.0283
aos  AP:65.46, 59.03, 52.78
2.
 INFO  Car AP@0.70, 0.70, 0.70:
bbox AP:25.2931, 35.8688, 30.2304
bev  AP:34.3534, 41.2079, 37.0070
3d   AP:13.5977, 17.5404, 14.6623
aos  AP:22.00, 34.01, 28.79
[... 153 lines cut ...]
bbox AP:24.7021, 21.3426, 19.1736
bev  AP:41.4406, 33.7527, 31.0403
3d   AP:32.6915, 27.1192, 23.5528
aos  AP:24.31, 20.93, 18.78

Cyclist AP@0.50, 0.50, 0.50:
bbox AP:56.9337, 50.9681, 44.7280
bev  AP:46.0885, 44.7921, 39.6599
3d   AP:23.9402, 22.0751, 21.4132
aos  AP:56.83, 50.86, 44.63
Cyclist AP_R40@0.50, 0.50, 0.50:
bbox AP:55.3930, 49.5998, 44.1024
bev  AP:43.7378, 40.7672, 35.9730
3d   AP:17.9729, 17.2490, 15.0542
aos  AP:55.30, 49.50, 44.00
Cyclist AP@0.50, 0.25, 0.25:
bbox AP:56.9337, 50.9681, 44.7280
bev  AP:72.5824, 65.1337, 62.6713
3d   AP:67.8893, 61.3175, 54.7505
aos  AP:56.83, 50.86, 44.63
Cyclist AP_R40@0.50, 0.25, 0.25:
bbox AP:55.3930, 49.5998, 44.1024
bev  AP:73.7297, 67.1657, 60.6393
3d   AP:66.5659, 60.2604, 54.0202
aos  AP:55.30, 49.50, 44.00