tudelft-iv/view-of-delft-dataset #59: Ego Velocity calculation

lkh5kor, 2023-08-16:
Hello, 

I was trying to find the ego vehicle speed and steering angle for each frame based on the suggestion in the previous issue https://github.com/tudelft-iv/view-of-delft-dataset/issues/47 
As suggested, the odom to camera transformation was used with a frame rate of 10 hz i.e 100 ms. 
Attaching the code used for the computation. 
[image]

The velocity and steering angle plot over time for first 550 frames is attached. 
[image]

[image]
[image]
Can you kindly confirm if the computed velocity and yaw rate is correct. 
(The velocity is not smoothed, it will be the next step.)
Thanks for the support and looking forward for the reply.

closed 2023-08-16 by andraspalffy

lkh5kor, 2023-08-17:
Hi @andraspalffy ,

Thanks for the fast feedback :)

yanconglin, 2024-02-12:
hey Andy,
same question on velocity estimation. 
I did a simple test using the t_utm_camera transformations, however, there seems to be something wrong. 

1. The velocity estimation for the initial frames goes horribly wrong;
2. transformation matrix missing for frame 543;

[image]
[image]

Please refer to the screenshots for details.
A code snippet is also available.

```import os
import numpy as np
from typing import Optional, List
import open3d as o3d
from vod.configuration import KittiLocations
from vod.frame import FrameDataLoader, FrameTransformMatrix, FrameLabels, homogeneous_transformation, homogeneous_coordinates

class VOD_dataset():
    def __init__(self, root_dir):
        self.root_dir = root_dir
        self.camera_dir = os.path.join(self.root_dir, 'lidar', 'training', 'image_2')
        self.frame_list = [ idx.replace('.jpg', '')  for idx in os.listdir(os.path.join(self.camera_dir))]
        self.frame_list.sort()
        # print(self.frame_list[0:100])
        print('total nbr of frames: ', len(self.frame_list))

    def load_transforms(self, idx):
        # print('idx: ', idx, self.frame_list[idx])
        kitti_locations = KittiLocations(root_dir=self.root_dir,
                                        output_dir="",
                                        frame_set_path="",
                                        pred_dir="",
                                        )
        frame_data = FrameDataLoader(kitti_locations=kitti_locations, frame_number=self.frame_list[idx])
        transforms = FrameTransformMatrix(frame_data)

        return transforms

    def compute_ego_velocity(self, transforms1, transforms2):
        cam1_pos = homogeneous_transformation(np.array([[0,0,0,1]]), transforms1.t_utm_camera) # t_map_camera also works?
        cam2_pos = homogeneous_transformation(np.array([[0,0,0,1]]), transforms2.t_utm_camera)
        # print('t_map_camera1: ', transforms1.t_map_camera)
        # print('t_map_camera2: ', transforms2.t_map_camera)
        velocity = np.linalg.norm(cam2_pos - cam1_pos) / 0.1 # 10Hz
        return velocity

    def __len__(self):
        # print('total samples: ', len(self.frame_list))
        return len(self.frame_list)

    def __getitem__(self, idx):
        transforms1 = self.load_transforms(idx)
        if idx==self.__len__()-1:
            idx2 = idx-1
        else:
            idx2 = idx+1
        # print('idxs: ', idx, idx2)
        transforms2 = self.load_transforms(idx2)

        velocity = self.compute_ego_velocity(transforms1, transforms2)

        return velocity

if __name__ =='__main__':
    dataset = VOD_dataset(root_dir='/home/yancong/Desktop/IV/view_of_delft/view_of_delft_PUBLIC')
    # dataset = VOD_dataset(root_dir='/pathto/view_of_delft/view_of_delft_PUBLIC')
    for idx in range(0, dataset.__len__()):
    # for idx in range(0, 10):
        velocity = dataset.__getitem__(idx)
        print(f'frame idx: {idx}, velocity: {velocity} m/s')

```