tudelft-iv/view-of-delft-dataset #61: How to use the evaluation function

min-zhang, 2023-08-18:
I used openpcd to reproduce the code about Radar, and the result was the evaluation result of openpcd, I don't know how to use your evaluation function in the code of openpcd, is there a specific step process to modify the code to get the result in your paper, including two areas.I read your tutorial and still don't know how to use it.
2023-07-05 17:30:01,903   INFO  Car AP@0.70, 0.70, 0.70:
bbox AP:38.2754, 43.7937, 33.9468
bev  AP:38.4106, 46.0381, 39.4115
3d   AP:21.5922, 228887, 21.5489
aos  AP:35.90, 39.16, 36.63
Car AP_R40@0.70, 0.70, 0.70:
bbox AP:36.3670, 37.6987, 32.4077
bev  AP:35.5528, 43.5953, 36.7469
3d   AP:15.8746, 19.0044, 17.7312
aos  AP:34.75, 37.00, 30.84
Car AP@0.70, 0.50, 0.50:
bbox AP:38.2754, 42.7937, 34.9468
bev  AP:55.4508, 57.8449, 50.1450
3d   AP:42.2968, 46.3682, 39.5512
aos  AP:35.90, 39.16, 34.63
Car AP_R40@0.70, 0.50, 0.50:
bbox AP:36.3670, 38.6987, 32.4077
bev  AP:54.1486, 57.2201, 48.2378
3d   AP:39.6693, 52.8487, 44.9838
aos  AP:41.75, 37.00, 37.84
Pedestrian AP@0.50, 0.50, 0.50:
bbox AP:28.6086, 27.1331, 26.0896
bev  AP:26.8249, 21.8494, 20.8565
3d   AP:12.8029, 11.0226, 10.2201
aos  AP:27.50, 25.83, 24.77
Pedestrian AP_R40@0.50, 0.50, 0.50:
bbox AP:26.4394, 22.8352, 20.6211
bev  AP:23.0455, 18.6491, 16.5344
3d   AP:9.3486, 8.1428, 6.9505
aos  AP:25.77, 22.05, 19.89
Pedestrian AP@0.50, 0.25, 0.25:
bbox AP:28.6086, 27.1331, 26.0896
bev  AP:48.5453, 41.1766, 39.4566
3d   AP:39.2645, 36.5330, 30.8388
aos  AP:27.50, 25.83, 24.77
Pedestrian AP_R40@0.50, 0.25, 0.25:
bbox AP:26.4394, 22.8352, 20.6211
bev  AP:47.7989, 39.7481, 35.1812
3d   AP:38.2997, 32.1196, 28.0489
aos  AP:25.77, 22.05, 19.89
Cyclist AP@0.50, 0.50, 0.50:
bbox AP:63.7330, 61.2801, 54.3684
bev  AP:64.2002, 62.0600, 55.1477
3d   AP:39.1116, 37.5996, 32.8188
aos  AP:63.59, 61.12, 54.23
Cyclist AP_R40@0.50, 0.50, 0.50:
bbox AP:65.6098, 59.1844, 52.9149
bev  AP:66.2757, 60.1361, 53.6604
3d   AP:36.5646, 33.4715, 29.9994
aos  AP:65.46, 59.03, 52.78
Cyclist AP@0.50, 0.25, 0.25:
bbox AP:63.7330, 61.2801, 54.3684
bev  AP:82.9964, 75.8251, 67.9316
3d   AP:75.4105, 73.1937, 65.3817
aos  AP:63.59, 61.12, 54.23
Cyclist AP_R40@0.50, 0.25, 0.25:
bbox AP:65.6098, 59.1844, 52.9149
bev  AP:82.9980, 76.1704, 69.0699
3d   AP:78.9620, 71.9490, 65.0283
aos  AP:65.46, 59.03, 52.78

andraspalffy, 2023-08-21:
Please see this.
[https://github.com/tudelft-iv/view-of-delft-dataset/issues/8#issuecomment-1231418778](https://github.com/tudelft-iv/view-of-delft-dataset/issues/8#issuecomment-1231418778)

closed 2023-08-21 by andraspalffy

min-zhang, 2023-08-23:
> Please see this. [#8 (comment)](https://github.com/tudelft-iv/view-of-delft-dataset/issues/8#issuecomment-1231418778)

I looked at the evaluation code you gave, but I don't know how to use it, do I need to replace the evaluation code of OpenPCD with the evaluation code under your VOD folder

andraspalffy, 2023-08-25:
Check this:

[https://github.com/tudelft-iv/view-of-delft-dataset/blob/main/4_evaluation.ipynb](https://github.com/tudelft-iv/view-of-delft-dataset/blob/main/4_evaluation.ipynb)

You can use it separately from OpenPCDet, just get your detections in a kitti format and feed it into to the given eval code.

shuxiusuxiu, 2023-09-09:
Hi！ @min-zhang 
i am wondering how you reproduced the radar version of the pointpillar. Did you just follow the guideline or do some modification?

cause i followed the guideline step by step, but cannot such high performance as you presented.

here are my results:

Car AP@0.70, 0.70, 0.70:
bbox AP:27.8012, 37.4001, 31.7605
bev  AP:37.1609, 45.7303, 38.8881
3d   AP:14.6725, 19.4797, 16.9983
aos  AP:24.09, 35.17, 29.88
Car AP_R40@0.70, 0.70, 0.70:
bbox AP:23.8732, 34.3879, 28.2622
bev  AP:34.9180, 43.0068, 36.3362
3d   AP:8.6750, 13.6588, 11.4928
aos  AP:19.97, 31.95, 26.27
Car AP@0.70, 0.50, 0.50:
bbox AP:27.8012, 37.4001, 31.7605
bev  AP:54.5013, 58.0862, 50.3914
3d   AP:38.8776, 46.5364, 39.9188
aos  AP:24.09, 35.17, 29.88
Car AP_R40@0.70, 0.50, 0.50:
bbox AP:23.8732, 34.3879, 28.2622
bev  AP:53.7749, 57.3217, 48.2844
3d   AP:36.5164, 44.1401, 37.1166
aos  AP:19.97, 31.95, 26.27
Pedestrian AP@0.50, 0.50, 0.50:
bbox AP:29.7623, 27.3935, 22.9308
bev  AP:25.9339, 21.3504, 20.3839
3d   AP:15.2857, 13.2255, 12.8167
aos  AP:29.22, 26.80, 22.54
Pedestrian AP_R40@0.50, 0.50, 0.50:
bbox AP:27.5007, 22.6826, 20.3741
bev  AP:21.2296, 17.8331, 15.8383
3d   AP:8.7615, 6.9108, 6.4170
aos  AP:26.82, 22.10, 19.82
Pedestrian AP@0.50, 0.25, 0.25:
bbox AP:29.7623, 27.3935, 22.9308
bev  AP:47.2477, 39.5948, 33.6610
3d   AP:39.2105, 31.8771, 30.4451
aos  AP:29.22, 26.80, 22.54
Pedestrian AP_R40@0.50, 0.25, 0.25:
bbox AP:27.5007, 22.6826, 20.3741
bev  AP:44.0939, 37.1022, 32.7399
3d   AP:36.9270, 30.4957, 26.6205
aos  AP:26.82, 22.10, 19.82
Cyclist AP@0.50, 0.50, 0.50:
bbox AP:62.0466, 55.3158, 53.4084
bev  AP:62.9435, 61.0005, 54.2983
3d   AP:31.4715, 30.9254, 29.5555
aos  AP:61.90, 55.12, 53.18
Cyclist AP_R40@0.50, 0.50, 0.50:
bbox AP:62.2394, 56.3321, 51.6059
bev  AP:63.1835, 58.6631, 52.5496
3d   AP:28.7065, 27.1802, 24.1858
aos  AP:62.08, 56.10, 51.35
Cyclist AP@0.50, 0.25, 0.25:
bbox AP:62.0466, 55.3158, 53.4084
bev  AP:83.6542, 76.0076, 68.0889
3d   AP:76.5496, 74.4370, 66.5705
aos  AP:61.90, 55.12, 53.18
Cyclist AP_R40@0.50, 0.25, 0.25:
bbox AP:62.2394, 56.3321, 51.6059
bev  AP:83.4082, 76.4154, 69.2619
3d   AP:80.2249, 73.2156, 66.1221
aos  AP:62.08, 56.10, 51.35