tudelft-iv/view-of-delft-dataset #62: Vehicle's Trajectory Reconstruction

paaraujo, 2023-08-21:
Hi,

I am traying to understand some specifics about the transformations available within the dataset and the toolkit. Could you please provide an example showing how to recover the vehicle's trajectory using such transformations/poses?

Thank you, Paulo.

andraspalffy, 2023-08-21:
Please see #59.

closed 2023-08-21 by andraspalffy

paaraujo, 2023-08-21:
Hi @andraspalffy , could you confirm that the data available is not a single trajectory? I am testing the code you suggested, and it seems that the trajectory returns to the origin from time to time. It seems that you stacked different segments. If I read the whole training sequence to recover the trajectory, the result doesn't seem realistic considering the dynamics of a land vehicle:

[image]

andraspalffy, 2023-08-22:
Dear @paaraujo, of course they are different segments. Please have a look at the images synced to the position files and you will see the jumps.
Andras