tudelft-iv/view-of-delft-dataset #77: 3d bounding box projection errors

UmangBDhanani, 2023-10-26:
When I am trying to project the 3d bounding boxes on the image, I am always getting a shift towards the sensor for every bounding box projection. As seen in the example images..

[image]
[image]

andraspalffy, 2023-10-26:
[https://github.com/tudelft-iv/view-of-delft-dataset/blob/main/3_2d_visualization.ipynb](https://github.com/tudelft-iv/view-of-delft-dataset/blob/main/3_2d_visualization.ipynb
)
We have ready to use visualization tools. What you probably miss is the misalignement between lidar and camera z axis,

explained here
[https://tudelft-iv.github.io/view-of-delft-dataset/docs/ANNOTATION.html#kitti-formatted-labels](https://tudelft-iv.github.io/view-of-delft-dataset/docs/ANNOTATION.html#kitti-formatted-labels)

closed 2023-10-26 by andraspalffy

UmangBDhanani, 2023-10-26:
Thank you @andraspalffy  for your reply. However I was mimicing the same visualization code into my own project but was missing a bit. Finally I have made it thanks.