tudelft-iv/view-of-delft-dataset #82: Bugs(?) in Evaluation

eat-slim, 2023-12-17:
Thanks for the excellent repo!

When I was testing the evaluation tool in ```4_evaluation.ipynb```, I took out the first 30 annotation files from ```view_of_delft_PUBLIC/lidar/training/label_2``` and placed them in a folder called ```test```. 
Next, I ran the following code:
```
evaluation = Evaluation(test_annotation_file=r'test')
results = evaluation.evaluate(
    result_path=r'test',
    current_class=[0, 1, 2])
```
annotation files and result files  are the same, so I think the result mAP should be 100. But I got the following results:
```
Results: 
Entire annotated area: 
Car: 0.0 
Pedestrian: 28.121212121212118 
Cyclist: 0.0 
mAP: 9.373737373737372 
Driving corridor area: 
Car: 0.0 
Pedestrian: 26.692456479690524 
Cyclist: 0.0 
mAP: 8.897485493230175 
```
Later, I tried to locate the reason for this result and found that when using ```instance_iou_eval``` to calculate IoU, for the same bbox_3d input, but it outputs a very low IoU (approximately 0.3). I wonder if there are any bugs in this function? -> ```vod.evaluation.rotate_iou_cpu.instance_iou_eval```

t-zhan, 2024-02-21:
> Thanks for the excellent repo!
>
> When I was testing the evaluation tool in `4_evaluation.ipynb`, I took out the first 30 annotation files from `view_of_delft_PUBLIC/lidar/training/label_2` and placed [...]

How do you evaluate in the end?

andraspalffy, 2024-03-08:
This is from the underlying code:

https://github.com/traveller59/kitti-object-eval-python/issues/4

TLDR some offset in angle between GT and detection is needed.

closed 2024-03-08 by andraspalffy

Petros626, 2025-05-09:
@andraspalffy why this fix isn't considered in `VoD`?

andraspalffy, 2025-05-10:
> [@andraspalffy](https://github.com/andraspalffy) why this fix isn't considered in `VoD`?

Beacause it fixes a case that basically will not happen in real application (zero rotation error), and we would diverge from the original codebase. Furthermore, if you insist, the solution is given:
```

eps=0.0001
return abab >= abap-eps and abap >= 0-eps and adad >= adap-eps and adap >= 0-eps
```

at https://github.com/tudelft-iv/view-of-delft-dataset/blob/a9df89290f118965b73e5cb4aa5409f42862894b/vod/evaluation/rotate_iou_cpu.py#L178