tudelft-iv/view-of-delft-dataset #85: Possibly incorrect labeling

lbhwyy, 2024-02-25:
I've noticed some errors in the labeling of bicycles and cyclists in the dataset, particularly with moving bicycles. There are two sequences in the dataset involving moving bicycles, from frame 4908 to 4943 and from frame 7371 to 7444. For example, in frame 4943, the label 'bicycle 1 0 1.2255948816291946 1422.9723 690.3846 1682.6979 1116.277 1.9313287528152072 0.6366684249736102 2.0659173025477258 3.130064186610863 2.3333640318643027 8.141027331184786 -4.6905343213758535 1' should instead be labeled as 'cyclist.' Similarly, in frame 7400, the label 'bicycle 1 0 1.7326526626443315 486.02783 687.54016 591.29865 867.14075 1.6192898190194869 0.7361264286612312 1.8092309769576511 -4.2281122664679245 2.2903098789347642 15.078906287113284 1.459273880970529 1' should also be labeled as 'cyclist'.

andraspalffy, 2024-04-28:
Thank you for the comment. Are the mistakes consistent, i.e. the cyclist is a ""bike"" across the ~40-70 frames?
can you add some pictures? Did you happen to correct it locally? Would be happy to release it as a fix.

andraspalffy, 2025-03-25:
Closing for inactivity

closed 2025-03-25 by andraspalffy