tudelft-iv/view-of-delft-dataset #86: A weird transformation in get_transformed_3d_label_corners in vod.visualization.helpers.py

t-zhan, 2024-03-12:
We found an unusual transformation in line 80 in vod.visualization.helpers.py:

center = (np.linalg.inv(t_camera_lidar) @ np.array([label['x'], label['y'], label['z'], 1]))[:3]

Does it mean that before I input labels to my model, I have to do the same transformation for the trainning labels? 

If so, do I need to do that transformation for GT labels before evaluation?

Toytiny, 2024-03-18:
please refer to another [repository](https://github.com/Toytiny/CMFlow) using the VoD dataset.

andraspalffy, 2024-03-23:
As explained here:
https://tudelft-iv.github.io/view-of-delft-dataset/docs/ANNOTATION.html#kitti-formatted-labels

We follow KITTI convention, thus labels are in camera frame both for training and for evaluation. If you use our eval code, you do not need to transform the GT labels, but you have to make sure that your code/framework outputs them in camera frame. In the latter I cannot help for every project unfortunately.

closed 2024-03-23 by andraspalffy

t-zhan, 2024-04-24:
> As explained here: https://tudelft-iv.github.io/view-of-delft-dataset/docs/ANNOTATION.html#kitti-formatted-labels
>
> We follow KITTI convention, thus labels are in camera frame both for training and [...]

In the original paper you implement the experiments on  OpenPCDet, a point-based object detection task. Which frame do you use in that implementation? If the predicted 3D bboxes are generated in radar frame, do you need to transform the GT labels from camera frame to radar frame?

t-zhan, 2024-04-24:
> > As explained here: https://tudelft-iv.github.io/view-of-delft-dataset/docs/ANNOTATION.html#kitti-formatted-labels
> > We follow KITTI convention, thus labels are in camera frame both for training and for evaluation. If you use our eval code, you do not need to transform the GT labels, but you have to make sure that your code/framework outputs them in camera frame. In the latter I cannot help for every project unfortunately.
> 
> In the original paper you implemented a point-based object detection task on OpenPCDet. Which frame do you use in that implementation? If the predicted 3D bboxes are generated in radar frame, do you need to transform the GT labels from camera frame to radar frame?

Is that project available for a reference? I wonder how to transform data and perform test.