tudelft-iv/view-of-delft-dataset #87: LiDAR Data Points not consistent

Michael-Schulze, 2024-03-18:
The LiDAR Data Points seems to be corrupted / wrong synchronized. In approx every 10 .. up to 30 frame the point cloud has "wholes" and it seems that some of the LiDAR points are using wrong height information.

You can see this behaviour e.g. in frame id 03285.

[image]

[image]

[image]

andraspalffy, 2024-03-18:
That is interesting, did not really see that before. Do you see it at other frames too? It could be noisy sensor data, or some kind of float issue.

Michael-Schulze, 2024-03-18:
It happens often, e.g. in frame ids:

03285
03306
03327
03342

So approx every 15 to 21 frame.

andraspalffy, 2024-04-13:
Thanks for looking into this. After having a look at KITTI which has similar lidar, I think this is a sensor problem, with which we have to live with.

closed 2024-04-13 by andraspalffy