tudelft-iv/view-of-delft-dataset #92: Hi, I have filed multiple requirements for access of the dataset. But I did not receive any reply.

zhanghaozhe23, 2024-04-13:
Thank you for your great work, I have filed the requirement for access of the dataset for a long time. But I did not receive any reply.
My email is zhanghaozhe23@shu.edu.cn

andraspalffy, 2024-04-13:
I already sent it to this email, I sent it again now.

closed 2024-04-13 by andraspalffy

zhanghaozhe23, 2024-04-15:
I am really sorry that I could not receive your email maybe because of the problem of the interception of my school's email. Could you please send it to my third-party email address? I am really interested in your work, thank you!

zhanghaozhe23, 2024-04-15:
> I already sent it to this email, I sent it again now.

I am really sorry that I could not receive your email maybe because of the problem of the interception of my school's email. Could you please send it to my third-party email address? I am really interested in your work, thank you!

andraspalffy, 2024-04-17:
We cannot send it to other email. Can you double check?
Write me an email at apalffy at tu delft nl