tudelft-iv/view-of-delft-dataset #96: Alignment of multi-frame radar dynamic object data

lotterf, 2024-05-08:
Hello, the data set published by your team is very helpful to me, thank you and your team for your free efforts. However, I still have some questions about the multi-frame radar in the data set. I found that the coordinate system has been calibrated and aligned by the self-driving sensor. However, how should I calibrate the speed of the dynamic objects in the scene? I think there might be some error in the data.

closed 2024-05-08 by andraspalffy

lotterf, 2024-05-08:
Yes, I know that radial velocity does not change after accumulation, but I would like to know if only radial velocity correction, instead of full velocity correction, will this lead to inaccurate multi-frame radar point cloud information after accumulation? For example, when a dynamic object is moving tangentially in its connection to a radar sensor. At this time, the radial velocity is 0, and there is a certain deviation in the image of the dynamic object between adjacent frames.

andraspalffy, 2024-05-09:
> I would like to know if only radial velocity correction, instead of full velocity correction, will this lead to inaccurate multi-frame radar point cloud information after accumulation?

Unfortunately I do not understand.