tudelft-iv/view-of-delft-dataset #105: Missing training labels

amirbakhtiary23, 2024-11-08:
Hi,
I've noticed several training labels are missing from the dataset e.g. scene 5098.
There are only 6,435 text files (label text files) at "view_of_delft_PUBLIC/lidar/training/label_2", whereas there are  8,682 scenes available. 
Am I missing something here? 
Regards,

andraspalffy, 2024-11-09:
As explained in the paper (and as done by the original KITTI), annotation of the test set is not published. The "missing" labels belong to the test frames, as indicated by the sets files in the IMAGESET folder. 

The testing folder is just a symlink of the training folder for your convenience.

closed 2024-11-09 by andraspalffy

amirbakhtiary23, 2024-11-10:
Oh sorry, totally forgot about that.
Thank you