tudelft-iv/view-of-delft-dataset #106: Few questions about the radar point clouds

amirbakhtiary23, 2024-12-03:
Hello,
Thank you for your great contribution. I had a few questions about radar point cloud and didn't want to bother you with email.
According to the paper : 
`The 3+1D radar outputs a point cloud with spatial, Doppler and reflectivity channels for each scan, giving a total of five features for each point: r range, α azimuth, θ elevation, vrel relative radial velocity, and RCS reflectivity. Since most point cloud based object detectors use Cartesian coordinates, we also transform the radar point cloud: p = [x, y, z, vrel, RCS],
where p denotes a point, and x, y, z are the three spatial coordinates with x and y axes pointing forward and left respectively w.r.t. the vehicle, see Figure 2. Compensated radial velocity is a signed scalar value denoted by vr, describing the ego-motion compensated (i.e. absolute) radial velocity of the point.`
Now the questions, if we consider these x,y,z as coordinates, which one would represent the depth? I suppose its x. 
And about the y,z. Since it's mentioned that Z represent elevation, i suppose it means the height, but how is height defined in these cases. I would appreciate if you could help me. Sorry if my questions are basic, I'm a CS major and don't know much about radars.

andraspalffy, 2025-03-25:
Please have a look here at the figure (also in the paper):
https://github.com/tudelft-iv/view-of-delft-dataset/blob/main/docs/SENSORS_AND_DATA.md

x, y, z are 3D cartesian coordinates, ortogonal to each other. There is no depth in this representation, to get that, you should calculate the (x^2 + y^2 + z^2)^0.5 range from the sensor. Z is not the elevation, but height.

closed 2025-03-25 by andraspalffy

amirbakhtiary23, 2025-04-08:
Thank you for taking the time to answer my question. 
A follow-up question though. If no depth is directly provided, why does [this](https://github.com/tudelft-iv/view-of-delft-dataset/blob/main/vod/frame/transformations.py#L361) piece of code use a variable named point depth and directly return its value without any further computation? 
`def project_pcl_to_image(point_cloud, t_camera_pcl, camera_projection_matrix, image_shape):
    """
A helper function which projects a point clouds specific to the dataset to the camera image frame.
    :param point_cloud: Point cloud to be projected.
    :param t_camera_pcl: Transformation from the pcl frame to the camera frame.
    :param camera_projection_matrix: The 4x4 camera projection matrix.
    :param image_shape: Size of the camera image.
    :return: Projected points, and the depth of each point.
    """
    point_homo = np.hstack((point_cloud[:, :3],
                            np.ones((point_cloud.shape[0], 1),
                                    dtype=np.float32)))

    radar_points_camera_frame = homogeneous_transformation(point_homo,
                                                           transform=t_camera_pcl)

    point_depth = radar_points_camera_frame[:, 2]

    uvs = project_3d_to_2d(points=radar_points_camera_frame,
                           projection_matrix=camera_projection_matrix)

    filtered_idx = canvas_crop(points=uvs,
                               image_size=image_shape,
                               points_depth=point_depth)
    uvs = uvs[filtered_idx]
    point_depth = point_depth[filtered_idx]

    return uvs, point_depth`

inputs are the raw radar point clouds

andraspalffy, 2025-04-09:
After projecting to camera frame, the first two coordinates, x and y relates to u v pixel positions, and the third one is the distance from the camera along the z axis, often called depth. 
This is only true in the camera frame though.