tudelft-iv/view-of-delft-dataset #107: Sequence information of differenet scenes

leehungu, 2024-12-10:
Hello, thanks for your nice work. I notice the dataset consists of several different scenes,  and I try to use this sequence information of differenet scenes. Could your provide the sequence information of differenet scenes? Such as the first scene is from 00000 to 00543. Thank you very much.

andraspalffy, 2025-03-25:
Hi, sorry for the delayed answer. Would this be useful?
|          | Start index in VoD | End index in VoD |
|----------|--------------------|------------------|
| delft_1  |                  0 |              543 |
| delft_2  |                544 |             1311 |
| delft_3  |               1312 |             1802 |
| delft_4  |               1803 |             2200 |
| delft_5  |               2201 |             2699 |
| delft_6  |               2700 |             3032 |
| delft_7  |               3033 |             3299 |
| delft_8  |               3300 |             3780 |
| delft_9  |               3781 |             4078 |
| delft_10 |               4079 |             4114 |
| delft_11 |               4115 |             4553 |
| delft_12 |               4554 |             4891 |
| delft_13 |               4892 |             5156 |
| delft_14 |               5157 |             5590 |
| delft_15 |               5591 |             6838 |
| delft_16 |               6839 |             7076 |
| delft_17 |               7077 |             7554 |
| delft_18 |               7555 |             7743 |
| delft_19 |               7744 |             8527 |
| delft_20 |               8528 |             8885 |
| delft_21 |               8886 |             9184 |
| delft_22 |               9185 |             9467 |
| delft_23 |               9468 |             9735 |
| delft_24 |               9736 |            10082 |
| delft_25 |              10083 |            10504 |
| delft_26 |              10505 |            10762 |
| delft_27 |              10763 |            10917 |

closed 2025-03-25 by andraspalffy

andraspalffy, 2025-09-06:
An update on this - in this version you see the frames RENUMBERED, and can see which split a certain scene belongs to. Also note: Scene 15 is in NEITHER for data quality reasons.

Number | Clip | Frames | Index Start Frame | Index End Frame | Split
-- | -- | -- | -- | -- | --
1 | delft_1 | 544 | 0 | 543 | Valid
2 | delft_2 | 768 | 544 | 1311 | Train
3 | delft_3 | 491 | 1312 | 1802 | Train
4 | delft_4 | 397 | 1803 | 2199 | Train
6 | delft_6 | 332 | 2200 | 2531 | Train
7 | delft_7 | 266 | 2532 | 2797 | Test
8 | delft_8 | 479 | 2798 | 3276 | Test
9 | delft_9 | 298 | 3277 | 3574 | Train
10 | delft_10 | 35 | 3575 | 3609 | Valid
11 | delft_11 | 438 | 3610 | 4047 | Train
12 | delft_12 | 338 | 4049 | 4386 | Train
13 | delft_13 | 265 | 4387 | 4651 | Train
14 | delft_14 | 434 | 4652 | 5085 | Valid
16 | delft_16 | 237 | 6334 | 6570 | Test
18 | delft_18 | 188 | 6571 | 6758 | Test
19 | delft_19 | 784 | 6759 | 7542 | Train
20 | delft_20 | 357 | 7543 | 7899 | Test
21 | delft_21 | 298 | 7900 | 8197 | Test
22 | delft_22 | 283 | 8198 | 8480 | Valid
23 | delft_23 | 268 | 8481 | 8748 | Train
24 | delft_24 | 347 | 8749 | 9095 | Train
25 | delft_25 | 422 | 9096 | 9517 | Test
26 | delft_26 | 258 | 9518 | 9775 | Train
27 | delft_27 | 155 | 9776 | 9930 | Train