tudelft-iv/view-of-delft-dataset #111: Help me 😭

skymun471, 2025-02-15:
Only the Car class shows a significant AP improvement, while Pedestrian and Cyclist do not yield meaningful AP results.

Do you have any insights into why this might be happening? 😭

INFO Car AP@0.70, 0.70, 0.70:
bbox AP:38.0021, 38.9439, 34.6891
bev AP:39.4407, 43.8751, 37.0377
3d AP:16.5685, 18.6300, 16.1584
aos AP:34.22, 37.36, 33.17
Car AP_R40@0.70, 0.70, 0.70:
bbox AP:36.3104, 37.4235, 31.1577
bev AP:36.2990, 41.2313, 34.5775
3d AP:10.7053, 12.6511, 10.7577
aos AP:32.09, 35.61, 29.59
Car AP@0.70, 0.50, 0.50:
bbox AP:38.0021, 38.9439, 34.6891
bev AP:55.7099, 57.2972, 49.3167
3d AP:42.2237, 45.0063, 39.1134
aos AP:34.22, 37.36, 33.17
Car AP_R40@0.70, 0.50, 0.50:
bbox AP:36.3104, 37.4235, 31.1577
bev AP:54.7911, 56.2914, 47.3264
3d AP:40.5273, 43.8209, 36.0583
aos AP:32.09, 35.61, 29.59
Pedestrian AP@0.50, 0.50, 0.50:
bbox AP:0.0399, 0.1186, 0.1186
bev AP:0.0297, 0.0297, 0.0297
3d AP:0.0230, 0.0230, 0.0230
aos AP:0.02, 0.06, 0.06
Pedestrian AP_R40@0.50, 0.50, 0.50:
bbox AP:0.0110, 0.0326, 0.0326
bev AP:0.0082, 0.0082, 0.0082
3d AP:0.0063, 0.0063, 0.0063
aos AP:0.00, 0.01, 0.01
Pedestrian AP@0.50, 0.25, 0.25:
bbox AP:0.0399, 0.1186, 0.1186
bev AP:0.0477, 0.0595, 0.0595
3d AP:0.0393, 0.0491, 0.0491
aos AP:0.02, 0.06, 0.06
Pedestrian AP_R40@0.50, 0.25, 0.25:
bbox AP:0.0110, 0.0326, 0.0326
bev AP:0.0131, 0.0164, 0.0164
3d AP:0.0108, 0.0135, 0.0135
aos AP:0.00, 0.01, 0.01
Cyclist AP@0.50, 0.50, 0.50:
bbox AP:0.0290, 0.0290, 0.0290
bev AP:0.0285, 0.0285, 0.0285
3d AP:0.0000, 0.0000, 0.0000
aos AP:0.03, 0.03, 0.03
Cyclist AP_R40@0.50, 0.50, 0.50:
bbox AP:0.0000, 0.0000, 0.0000
bev AP:0.0000, 0.0000, 0.0000
3d AP:0.0000, 0.0000, 0.0000
aos AP:0.00, 0.00, 0.00
Cyclist AP@0.50, 0.25, 0.25:
bbox AP:0.0290, 0.0290, 0.0290
bev AP:0.0441, 0.0441, 0.0441
3d AP:0.0397, 0.0397, 0.0397
aos AP:0.03, 0.03, 0.03
Cyclist AP_R40@0.50, 0.25, 0.25:
bbox AP:0.0000, 0.0000, 0.0000
bev AP:0.0000, 0.0000, 0.0000
3d AP:0.0000, 0.0000, 0.0000
aos AP:0.00, 0.00, 0.00

Petros626, 2025-03-15:
I think this repo is dead.

andraspalffy, 2025-03-25:
The repo is not dead, but unfortunately we do not have the capacity to provide individual advice. The original poster did not mention what method are they training/testing and how, which makes it very hard to comment. Please have a look here, maybe it helps:

https://github.com/tudelft-iv/view-of-delft-dataset/blob/main/PP-Radar.md

closed 2025-03-25 by andraspalffy