tudelft-iv/view-of-delft-dataset #113: Evaluation with a PyTorch model from another framework?

Petros626, 2025-03-15:
Hello guys,

awesome work!

Is it possible to use a `model.pt` from my framework and evaluate it with my predictions and gt-labels?

andraspalffy, 2025-03-25:
If you run your inference in your own framework, you can export the detections. Once you convert that into the format of the VoD dataset (very similar to KITTI) then you can evaluate them.

closed 2025-03-25 by andraspalffy

Petros626, 2025-03-25:
When I provide for the detections only rotated 2D BBoxes from the BEV perspective, how can the metrics for bbox(2D I guess) , 3D calculated? I believe that then only bev AP would be possible.

andraspalffy, 2025-03-25:
Without knowing the specifics of your setup, I think you are right, BEV AP is the only reasonable one. You can try 3D by assigning a "fake" height to all boxes (e.g. bike = 1.7 m tall) and vertical offset (e.g. every box is on the plane = -1 m or so), but will not be very meaningful.

Petros626, 2025-03-25:
Thanks for the clarification Andras, coming back to you in case of further issues.

Petros626, 2025-04-15:
Hey @andraspalffy,

I'm trying now to validate my bird's eye view dataset based on `KITTI`. For the training framework I was forced to convert h, w, l (in m) to 640x640 bev-image space and further to normalize the coordinates [0...1].

How can I reverse this procedure for my gt-data and the saved predictions?