tudelft-iv/view-of-delft-dataset #115: my radar result is very confused

ding1007, 2025-03-29:
hello,I follow the PP-radar.md, and follow the option 1 (comment the truncation) this is my option 1 result.( https://github.com/tudelft-iv/view-of-delft-dataset/issues/8#issuecomment-1172435763)
2025-03-02 13:26:23,423   INFO  Car AP@0.70, 0.70, 0.70:
bbox AP:25.7179, 35.4292, 30.5501
bev  AP:34.1290, 41.1492, 37.2577
3d   AP:16.1179, 20.0499, 17.3645
aos  AP:23.17, 33.81, 29.22
Car AP_R40@0.70, 0.70, 0.70:
bbox AP:22.1645, 32.9946, 27.4748
bev  AP:31.3542, 40.2137, 34.8378
3d   AP:10.6294, 15.7049, 12.7459
aos  AP:19.40, 31.22, 26.01
Car AP@0.70, 0.50, 0.50:
bbox AP:25.7179, 35.4292, 30.5501
bev  AP:50.3347, 56.4008, 49.0762
3d   AP:33.5560, 40.7543, 36.8160
aos  AP:23.17, 33.81, 29.22
Car AP_R40@0.70, 0.50, 0.50:
bbox AP:22.1645, 32.9946, 27.4748
bev  AP:49.5033, 55.4026, 48.0839
3d   AP:30.4313, 39.8035, 33.4949
aos  AP:19.40, 31.22, 26.01
Pedestrian AP@0.50, 0.50, 0.50:
bbox AP:30.6035, 28.3279, 23.6951
bev  AP:25.9212, 21.0859, 20.0875
3d   AP:15.1936, 11.0828, 10.7266
aos  AP:30.18, 27.73, 23.29
Pedestrian AP_R40@0.50, 0.50, 0.50:
bbox AP:28.5505, 24.4241, 20.9913
bev  AP:23.2335, 18.8710, 16.8119
3d   AP:11.7064, 8.9852, 7.6768
aos  AP:27.97, 23.69, 20.38
Pedestrian AP@0.50, 0.25, 0.25:
bbox AP:30.6035, 28.3279, 23.6951
bev  AP:47.1639, 40.5403, 33.9350
3d   AP:40.0421, 32.5844, 31.2589
aos  AP:30.18, 27.73, 23.29
Pedestrian AP_R40@0.50, 0.25, 0.25:
bbox AP:28.5505, 24.4241, 20.9913
bev  AP:44.2866, 37.5757, 33.1843
3d   AP:37.9014, 31.3763, 27.3054
aos  AP:27.97, 23.69, 20.38
Cyclist AP@0.50, 0.50, 0.50:
bbox AP:70.5925, 63.4788, 56.1974
bev  AP:69.3905, 62.0143, 55.3953
3d   AP:38.8189, 37.3343, 32.8143
aos  AP:70.47, 63.32, 56.05
Cyclist AP_R40@0.50, 0.50, 0.50:
bbox AP:70.8167, 62.5826, 57.5044
bev  AP:68.0069, 61.4810, 55.0312
3d   AP:36.4647, 32.9068, 29.5099
aos  AP:70.68, 62.42, 57.32
Cyclist AP@0.50, 0.25, 0.25:
bbox AP:70.5925, 63.4788, 56.1974
bev  AP:83.6338, 75.9173, 67.8635
3d   AP:81.8139, 73.7774, 66.3011
aos  AP:70.47, 63.32, 56.05
Cyclist AP_R40@0.50, 0.25, 0.25:
bbox AP:70.8167, 62.5826, 57.5044
bev  AP:85.0083, 77.8516, 70.6264
3d   AP:81.6989, 72.7935, 65.7329
aos  AP:70.68, 62.42, 57.32
you said HARD version is similar to your paper result (https://github.com/tudelft-iv/view-of-delft-dataset/issues/8#issuecomment-1154851816) And I also convert the result.pkl to kitti format (00000.txt...), then use your evaluation function, this is my result very low ,so I am very confused what happened?
mAP 3D BBox finished
Evaluating kitti by ROI
mAP Image BBox finished
mAP bev BBox finished
mAP 3D BBox finished
Results: 
Entire annotated area: 
Car: 31.54360458761179 
Pedestrian: 0.1976284584980237 
Cyclist: 6.803651202372846 
mAP: 12.84829474949422 
Driving corridor area: 
Car: 66.67442341136086 
Pedestrian: 0.06939625260235947 
Cyclist: 10.80457502816017 
mAP: 25.849464897374464

ding1007, 2025-03-30:
I solve this problem,this is the result,dimension lwh is problem.mAP 3D BBox finished
Evaluating kitti by ROI
mAP Image BBox finished
mAP bev BBox finished
mAP 3D BBox finished
Results: 
Entire annotated area: 
Car: 36.816049586059464 
Pedestrian: 31.25894832634435 
Cyclist: 66.30110344208455 
mAP: 44.79203378482945 
Driving corridor area: 
Car: 70.07857624445806 
Pedestrian: 41.17232649605538 
Cyclist: 87.56152469252481 
mAP: 66.27080914434609

closed 2025-03-30 by ding1007

sleep-earlyyy, 2025-06-21:
> I solve this problem,this is the result,dimension lwh is problem.mAP 3D BBox finished Evaluating kitti by ROI mAP Image BBox finished mAP bev BBox finished mAP 3D BBox finished Results: Entire [...]

Hi, I wonder where did you change the lwh? The value of my framework evaluation is similar to yours, but the value of vod evaluation is much lower.