tudelft-iv/view-of-delft-dataset #117: dataset look incomplete

MarcusnHuang, 2025-04-23:
Thank you for your work, but why does the dataset I downloaded look incomplete, the `image_2, label_2, pose` in the `radar/train` folder all look problematic
[image]

Lucasmogsan, 2025-04-23:
> Thank you for your work, but why does the dataset I downloaded look incomplete, the `image_2, label_2, pose` in the `radar/train` folder all look problematic [image]

I had similar confusion but seems like they refer to other folders that contain e.g. camera information (which is in the lidar folder).

andraspalffy, 2025-04-27:
The dataset is not "incomplete". 

We wanted to follow the KITTI format, and as explained here: 

https://tudelft-iv.github.io/view-of-delft-dataset/docs/GETTING_STARTED.html 

that basically means we provide 4 "sub"dataset, each KITTI formatted, where the `velodyne` folder contains different point clouds (lidar, radar, radar 3, radar 5 scans). 

All the other data (camera, labels, pose) are identical, thus we used symlinks to save users from unneccessary data transfer.

Thanks @Lucasmogsan for explaining!

see also this issue: https://github.com/tudelft-iv/view-of-delft-dataset/issues/24

closed 2025-04-27 by andraspalffy

Petros626, 2025-05-02:
@andraspalffy when will our applications for the dataset be processed? related to https://github.com/tudelft-iv/view-of-delft-dataset/issues/114

MarcusnHuang, 2025-05-07:
> 数据集并非“不完整”。
> 
> 我们想遵循 KITTI 格式，如下所述：
> 
> https://tudelft-iv.github.io/view-of-delft-dataset/docs/GETTING_STARTED.html
> 
> 这基本上意味着我们提供 4 个“子”数据集，每个 KITTI 格式，其中文件夹包含不同的点云（激光雷达、雷达、雷达 3、雷达 5 扫描）。`velodyne`
> 
> 所有其他数据（相机、标签、姿势）都是相同的，因此我们使用符号链接来避免用户不必要的数据传输。
> 
> 谢谢你的解释！
> 
> 另见这个问题：[#24](https://github.com/tudelft-iv/view-of-delft-dataset/issues/24)

Thanks for your answer！