tudelft-iv/view-of-delft-dataset #120: Confusion about code using BEV evaluation

Petros626, 2025-05-06:
Hello guys,

through my transformation process from bev annotations back to 3d annotations I saw these line of code, which confused me:
https://github.com/tudelft-iv/view-of-delft-dataset/blob/a9df89290f118965b73e5cb4aa5409f42862894b/vod/evaluation/kitti_official_evaluate.py#L418 https://github.com/tudelft-iv/view-of-delft-dataset/blob/a9df89290f118965b73e5cb4aa5409f42862894b/vod/evaluation/kitti_official_evaluate.py#L420

Why from the label `x,y,z,h,w,l,rotation_y` only `x,z,h,l,rotation_y` should be taken for BEV evaluation? I thought `x,y,w,l,rotation_y` is required.

Thanks for any clarification!

Shiming94, 2025-05-22:
Hi Petros, 

The location was defined in the camera frame, so the `x` and `z` are actually used for BEV, and `y` is the vertical direction. That is my understanding.

Best,
Shiming

Petros626, 2025-06-12:
Yes it's `x`, `y` for the LiDAR frame and `x` and `z` for the Camera frame. Thank you!

Transformation solution here: https://github.com/jhultman/vision3d/issues/27#issuecomment-2884262931

closed 2025-06-12 by Petros626